supermerill/SuperSlicer #2: Gyroid infill set as bridge feature.

rongith, 2018-07-07:
### Version
version_1.40.1-rc-33-g9cb144dd

### Operating system type + version
Windows 10 x64 1803

### Behavior
Gyroid infill set as bridge feature.
[image]

#### STL/Config (.ZIP) where problem occurs
[04513-r01f - I3D-1.zip](https://github.com/supermerill/Slic3r/files/2172645/04513-r01f.-.I3D-1.zip)

supermerill, 2018-07-08:
I don't understand.

rongith, 2018-07-08:
With Prusa 1.40.1, the gyroid infill is "Internal infill" and shown as dark red color. With version_1.40.1-rc-33-g9cb144dd, the gyroid infill is all "Bridge infill" and shown as light blue color.
version_1.40.1-rc-33-g9cb144dd
[image]
[04513-r01f - I3D-1-20180708-191647-(P)04513-Light-MK3-(F)ColorFabb XT - chaud2-(P)PMK3-R0.8-M860-1.40.1+win64.zip.amf.zip](https://github.com/supermerill/Slic3r/files/2173942/04513-r01f.-.I3D-1-20180708-191647-.P.04513-Light-MK3-.F.ColorFabb.XT.-.chaud2-.P.PMK3-R0.8-M860-1.40.1.win64.zip.amf.zip)

supermerill, 2018-07-11:
reproduced, i will look at it

rongith, 2018-07-16:
Didn't know it was intentional to set gyroid infill as bridge. Indeed that way has its logic too although at the moment I prefer not having bridge infill for gyroid as the flow rate is usually not the same at all.
```
    // require bridge flow since most of this pattern hangs in air
    -virtual bool use_bridge_flow() const { return true; }
    +virtual bool use_bridge_flow() const { return false; }
```

supermerill, 2018-07-16:
I think personally that it's unnecessary (infill with bridge setting), as we can tune the infill speed anyway, an it's printing very (enough) well at high speed.

commit 411ceb76 (2019-12-11): fix #2 for ShortestPath

Wrong idx in reorder_extrusion_entities
closed 2018-08-03 by supermerill