supermerill/SuperSlicer #3: No "denser infill below top layers" when I define 2 perimeters.

rongith, 2018-07-07:
### Version
version_1.40.1-rc-33-g9cb144dd

### Operating system type + version
Windows 10 x64 1803

### Behavior
No "denser infill below top layers" when I define 2 perimeters. But there is "denser infill below top layers" when I define 3 perimeters.

#### STL/Config (.ZIP) where problem occurs

[04513-r01f - I3D-1-20180707-174902-(P)04513-Light-MK3-(F)ColorFabb XT - chaud2-(P)PMK3-R0.8-M860-1.40.1-rc.zip.amf.zip](https://github.com/supermerill/Slic3r/files/2172885/04513-r01f.-.I3D-1-20180707-174902-.P.04513-Light-MK3-.F.ColorFabb.XT.-.chaud2-.P.PMK3-R0.8-M860-1.40.1-rc.zip.amf.zip)

supermerill, 2018-07-11:
I have correct that by changing the way it work, now it's "denser infill below solid layers" and use a dense infill on all your part because of the small slope...
Should I stick to "below top layers" (but i will only work if there are enough perimeters), or should I try to split these surfaces if they are too big ?

rongith, 2018-07-16:
It is OK with 2 perimeters now. But the drawback of your modification is:
- it changes the infill rate on all the layer, not only under solid infill area
- as soon as the is solid infill, all the next layers are with denser infill rate regardless of the real necessity. 
The consequence for the joined case is that I have a denser density on le whole part as soon as the layer 16 (4.10mm height) due to small top solid areas and for all the next layers.
[04513-r01f - I3D-3-20180716-162127-(P)04513-Light-MK3-(F)ColorFabb XT - chaud2-(P)PMK3-R0.8-M860-1.40.1-rc.zip.amf.zip](https://github.com/supermerill/Slic3r/files/2198011/04513-r01f.-.I3D-3-20180716-162127-.P.04513-Light-MK3-.F.ColorFabb.XT.-.chaud2-.P.PMK3-R0.8-M860-1.40.1-rc.zip.amf.zip)
"below top layers" should be the way to go if the order is always "infill"-"solid infill"-"top solid infill". Don't know if there are cases that can be different.

supermerill, 2018-07-24:
I added a threshold: if the surface is at least 3 time larger than the supported part, it will be split.
Also The pattern is defaulted to rectilinear, and the angle can be chosen.

closed 2018-08-03 by supermerill