supermerill/SuperSlicer #5: Vase moed broken

supermerill, 2018-09-14:
When slicing in vase mode, the line are not joined and too many layer are created.

Probably something related to the "last layer at exact height" function
[image]

Murloc992, 2019-02-06:
This issue is still valid on 1.4.2 alpha 4.

I have no exact layer marked.

[image]

supermerill, 2019-02-14:
@Murloc992 : can you post your settings?

Murloc992, 2019-02-14:
@supermerill Sure.

Here you go:
[bugreport.zip](https://github.com/supermerill/Slic3r/files/2864549/bugreport.zip)

I will add benchy and all the other bug report info soon to their respective issues soon.

supermerill, 2019-02-14:
I corrected the bug. Note that you don't need to hollow the object to create a hollow cylinder.

closed 2019-02-21 by supermerill