supermerill/SuperSlicer #7: Supporting dense layer

rongith, 2018-10-01:
### Version
Slic3r++_1.41.2_win32_20180926

### Behavior
From https://github.com/supermerill/Slic3r/issues/3
"as soon as there is some solid infill, all the next layers are with denser infill rate regardless of the real necessity.
The consequence for the joined case is that I have a denser density on le whole part as soon as the layer 16 (4.10mm height) due to small top solid areas (layer 19 -5.0mm height) and for all the next layers."

That issue still occurs.

[image]

supermerill, 2018-10-01:
I need an advice: the density of the dense layer is on the new version fixed to 50%

Do you think i should add a choice for a 25% density?

rongith, 2018-10-01:
Don't know. Didn't test it for now due to that issue. For the part i use to print, the time to print increase from 8:40 up to 17:40 with "supporting dense layer" enabled. Then i tend to add top solid layers.
I read the wiki and as i understand there is only one layer concerned by denser infill and the modification is not for the full layer. Then my advice would be 50%.

supermerill, 2018-10-01:
I created a test build : https://mega.nz/#!plhiyaZY!KElMmgKYIk7AeI_fumxSnQ9PvSHsPwq_O67FK9AXiCg
Also available at the master branch.

It create a switch for the dense layer algo, the bad new one or the good old one.
I wonder if i should re-add the number of dense layers or if I can let it stay at 1.

rongith, 2018-10-01:
Test build of the link doesn't provide option to switch algorithm. Then i compiled from the master branch and succeeded as you added MedialAxis.cpp. Algorithm "Anchored" seems to do the job.
I will test it when finished my printing with Slic3r++_1.41.2_win32_20180926. I made some simulation and I fear there is not much to gain from additional layers.
With 10% infill. I tested with 4 top layers (not good) and 5 top layers (could be the value). I'm printing with 6 top layers. Then I will test 4 top layers and denser supporting layer.

supermerill, 2018-10-03:
> Test build of the link doesn't provide option to switch algorithm.

Oh right, i forgot to copy the dll....

> Algorithm "Anchored" seems to do the job.
> I fear there is not much to gain from additional layers.
>With 10% infill. I tested with 4 top layers (not good) and 5 top layers (could be the value). I'm printing with 6 top layers. Then I will test 4 top layers and denser supporting layer.

that means it's ok or I have to do some improvements?

closed 2018-12-07 by supermerill