supermerill/SuperSlicer #8: Merge with Slic3rPE 1.41.1

tmolberg, 2018-10-06:
Are you planning to merge with the latest release of Slic3rPE again? The new support blocker system seems to be very interesting and would be fantastic to use in combination with your features

### Version
_Version of Slic3r used goes here_

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_
Yes

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

supermerill, 2018-10-07:
yes, but not with the beta. I will merge when the release will be published.

supermerill, 2018-10-31:
done

closed 2018-10-31 by supermerill