supermerill/SuperSlicer #10: travel over perimeter

supermerill, 2018-11-20:
"avoid crossing perimeter" isn't  good enough to avoid oozing & [something]. We need a better option.

burner-, 2019-04-26:
I have suggestion to that. Currently there is zlift but it will left blobs and slow down print and cause even more quality problems, jumps, failed prints and it basically ruin your day. My suggestion is parabolic flight zlift for travel. Basically if z path is like this https://nc.fsol.fi/s/DrFjDptt2KbBrmC  it will cause wiping same time when x/y accelerate and z lifts smoothly. And when it land to end point of travel that angle is calculated so that Z will do maximum deacceleration at same time when X/Y will deaccelerate to wanted point. So that zlift will not slow down print by slowing down X/Y motion. I think that will resolve all problems what current zlift solution does and also fix quite lot of oozing, stringing and other problems what you get with large extruder.

supermerill, 2019-04-26:
It's not really a "avoid perimeter" option but a new "wipe" thing.
Why not, but will it work equally good for long and short travels? Last time i implemented a thing for someone, i didn't had any feedback after, if I implement that, will you dedicate some days to test it to see if it's useful or not?

burner-, 2019-04-27:
That is true. It is more like wipe&zlift at same. I think it will work also with long and short travels but probably that formula can need experimenting. There is one idea howto do that:
- max height
- lift ratio (so total x/y velocity will get calculated to zlift heigh)
That may or may not work it self. BUT it can be that it give even better result if lift come from ratio and landing is done by maximum decent rate what is possible when it approach end of travel. 

So that way if travel is short it will only lift little and if travel is long/high speed it will rise higher because risk to high speed collision is higher and also oozing is more significant at long travels. 
I hope that you understand even some part of my semiconscious mumble =)

And about testing. Yes I can test that. I have 360x530mm bend and currently printer can travel 300mm/s but I can run it only with 150mm/s because when that 1mm nozzle hits to blob with 300mm/s it break things and skip steps etc. 

I have currently about 7kg prints at print queue so I cant do tests with those.

dot-bob, 2019-04-27:
I will test this feature as well.  I typically pull and build from source whenever there are new commits.

burner-, 2019-04-28:
I dont have build environment by my self but is there good documentation for doing it or is there nightly builds?

supermerill, 2020-06-03:
Prusa has improved it already. Won't do anything more in the short term.

closed 2020-06-03 by supermerill