supermerill/SuperSlicer #13: Suporting dense layers with "Automatic" algorithm not working?

Sebastianv650, 2018-12-06:
I'm using 1.41.1-5 and playing around with the new options. When I enable "Suporting dense layers" with algorithm automatic, nothing changes before a solid layer. Only "Anchored" inserts a denser infill layer. Is this a bug or a misunderstanding of it's way of working?

Screenshot of infill with "Automatic" just before the first real solid layer:
[image]

supermerill, 2018-12-06:
I recently had the same problem. Maybe i added a switch to prevent the dense infill on "full" areas. I have to remove it.

supermerill, 2018-12-07:
should be fixed in 1.41.2-1

Sebastianv650, 2018-12-07:
It is, thanks!

closed 2018-12-07 by Sebastianv650