supermerill/SuperSlicer #14: Ironing, adjust flow factor possible?

Sebastianv650, 2018-12-07:
### Version
1.41.2

Using PLA and my Prusa i3Mk2, my flow rate is spot on for normal printing. However when I try the Ironing feature, some parts of the top surface are smooth and others are rough like sand paper. It seems as it would need a slight increase of the flow rate for ironing, is it possible to adjust this (guess not)?

I read in the wiki that you apply 90% flow rate at first pass and 10% on the second. So it should add up to 100%, but maybe due to the double pass it fills more small gaps that would be containing air with only 1 pass between the layers resulting in a more dense layer therefore needing some % more filament..?

supermerill, 2018-12-07:
Try to increase the over-bridge flow ratio, or/and use the "Suporting dense layer" option.

A quick check to see if it's the problem: print a flat disc (3-5 layer height, all 100% infill) and a cube (with sparse infill inside). If the top surface is good on the flat one and rough on the other, it's the issue.

Sebastianv650, 2018-12-07:
I already did this test, no change in the results.

Sebastianv650, 2018-12-08:
Here is my setup as 3mf, this is the ironing version. The non - ironing pic has only two changes:
- Top fill set to rectilinear
- 1 perimeter (by accident, shouldn't change anything)

The first layer is set to a quite dense infill but not 100% to avoid bed leveling influences propagating up into the top layer.
[20x20x10.txt](https://github.com/supermerill/Slic3r/files/2659516/20x20x10.txt) (Rename to .3mf)

One note about how I judge the esteps/mm good or bad. I'm varying the flow rate starting from "too low" and rising until no gap is visible. Then I'm switching to the microscope, searching for more gaps and feeling the surface with a thin needle. When it's possible to slide the needle tip from one line to the next without it falling in a "pit" between the lines, it's perfect.

Picture of normal top infill, complete and magnified:
[image]
[image]
[image]

Picture with ironing. It's not easy to see, but if you split the square into 4 smaller squares,  the one on the lower right is the rough part:
[image]
Rough part magnified, note the visible horizontal lines:
[image]
Better section magnified, more even look:
[image]

Sebastianv650, 2018-12-08:
I did some more tests, but all lead to the same result (1/4 of the square beeing rough)
- Switching to another place on the print bed
- Increasing the test piece height to 3mm
- Increasing the flow rate to 1.03
- Disabling linear advance

Sebastianv650, 2018-12-08:
Sorry for generating a "false issue".
In reality what happened here is that I used my TAZ 5 up to this day, the MK2 was nearly never running as I only had the one color of filament it came with up to this week. As I'm now switching to 1.75mm filament and therefore the MK2 is now running, I was sure that I calibrated the MK2 when it arrived a long time ago. But I didn't.
It was close, but it over-extruded some %. I calibrated it, and ironing works as intended now I guess. Using ironing, the surface becomes a little bit smoother and the typical visible 3D printing lines are nearly gone. But the downside, which might depend on the filament, is that with my red one the surface finish changes from a slight glossy to a dull one.

Therefore I'm closing this one.

closed 2018-12-08 by Sebastianv650