supermerill/SuperSlicer #16: Weird ironing behavior

Murloc992, 2018-12-12:
### Version
Slic3r++ 1.41.2-1 Win32 release

### Operating system type + version
Windows 10

### Behavior
Setting the attached STL to 100%, the generated preview and the print does some weird bridging moves while sanding on the same layer. Any ideas what can be done to fix this issue?

[image]

Layer that goes before ironing happens:
[image]

Layer where ironing happens:
[image]

How it looks after printing:
[image]

As you can see the weird bridge infill things are present..

#### STL/Config (.ZIP) where problem occurs
Config is default(prusa MK2.5) only with ironing set. Infill is set to 100%.

[fans.zip](https://github.com/supermerill/Slic3r/files/2672995/fans.zip)

Murloc992, 2018-12-12:
Interestingly enough, default Slic3r PE does not have these bridges at all..

[image]

Murloc992, 2018-12-12:
[3mf_file.zip](https://github.com/supermerill/Slic3r/files/2673104/3mf_file.zip)

Here you go. :)

Murloc992, 2018-12-12:
You also seem to have a '++' profile. Custom default settings?

supermerill, 2018-12-12:
Seems to be the "only one perimeter on top surface" that create that bug.
I'll look at it for the next release.
For the settings, I have many test settings and no "main" one.

Murloc992, 2018-12-12:
Yeah, that seems to be the issue. Thanks for looking into it. I love this version of Slic3r so much I wish I had found it sooner. :)

photog0411, 2018-12-15:
Hi, also query about the ironing feature.

What speed setting will apply to ironing?  I have my top solid infill set to 30mms speed, and my sparse infill set to 200mms speed, but looking at the gcode preview, it appears that the speed of 200mms is being applied to the top (ironing) layer. Is this supposed to be that way?

supermerill, 2018-12-15:
Ironing is done in two steps, one rought, one fine.
The first pass use the solid infill, the second one use the top solid infill.

photog0411, 2018-12-15:
> 
> 
> Ironing is done in two steps, one rought, one fine.
> The first pass use the solid infill, the second one use the top solid infill.

I guess maybe it's bug then in visual preview of gcode speed.

See here for example. I have solid infill set to 180mms, and top infill set to 30mms. But in gcode preview, it shows the top most infill as the red color for speed which is 180mms. Or am I reading it wrong and confused?

[image]
[image]

photog0411, 2018-12-15:
Ok thank you, I must have missed the smaller lines because hard to see.

One final question:  With Ironing should I be using a fast speed or slower speed? I'm never used cura (which I think this attempts to duplicate) so I'm not sure how it's supposed to work, and if your method is exactly same as cura.

I know with standard solid top infill (no ironing) generally we want slower speed for better detail, but I'm not sure if ironing is supposed to be fast.

supermerill, 2018-12-16:
I had good result with top infill speed at ~30mm/s (on my prusa mk2).

supermerill, 2019-02-14:
I fixed the bug. It was a very little area that was considered as unsupported, so it was grow to anchor it into the part. I added a check to remove any bridge infill if it's thinner than a single line.

closed 2019-02-21 by supermerill

LaserBee, 2019-02-22:
It works fantastic now, thank you very much!