supermerill/SuperSlicer #19: Ironing making holes in the top layer

rlesac, 2018-12-16:
Using 1.41.2-1 I get random places where ironing will "burn" through the top layer.
Do you have any suggestions on how to avoid it?
I changed the top solid speed from 40 to 60mms, and it improved things a bit.
I've set this print to be a bit overextruded and bumped the infill/perimeters overlap to 50% so there should be enough plastic to cover it nicely when ironed.

In Cura 60mms works fine for ironing.

Here are examples:
[image]
[image]
[image]

rlesac, 2018-12-17:
I apologize, I have somehow missed the wiki, and sure enough it deals with the aforementioned issue. Will try the print again tomorrow.

rlesac, 2018-12-17:
After some testing, these are my results:
The over-bridge flow ratio seems to have resolved the ironing issues over sparse infill.

I still have the issue when using thin models with 3 solid layers (0.6mm) and using ironing. 
I upped the flow a bit, but could not get a  good print with only 3 layers and ironing. 
I got around the issue by using 4 layers and that seems to help, but it's not ideal.

Cura ironing has no such issues.

Do you have any ideas or recommendations?

supermerill, 2018-12-17:
> Cura ironing has no such issues.

I didn't try their ironing. From what i have seen in reddit, some users have the same issue. (https://www.reddit.com/r/3Dprinting/comments/926nqe/cura_ironing_neosanding_issues_holes_scars/)

> Do you have any ideas or recommendations?

I need some pics of the remaining problems & .3mf to give a useful advice (plater->export as .3mf).

photog0411, 2018-12-19:
would creating a separate speed setting for ironing only be possible I wonder? For those of us that use a lot of profiles it might be helpful

supermerill, 2018-12-19:
It's possible. Maybe i can store the value for each pattern and change it on the fly in the gui.

closed 2019-01-23 by supermerill