supermerill/SuperSlicer #20: [Feature Request]  Zero Travel Mode (cookie cutter mode)

photog0411, 2018-12-19:
Maybe it's already possible some how but here's my issue.

I make cookie cutters. The top most layers all all just outside perimeters. But even with "avoid crossing perimeters" setting checked, it still do cross some perimeters. And I get strings. Check below photo for example.

I wonder if it's possible to do continuous looping perimeters with ZERO travel so I never get the stringing. Maybe useful for other things too 

[image]

supermerill, 2018-12-19:
You can play with the bottom panel in the layer tab.
Disable "thin walls" (to remove most of gap fill)

looping perimeter -> activate to remove moves between perimeters
seam-> nearest
travel moves reduced -> activate it

if multiple objects, use "complete individual objects" in sequential printing in output option tab.

photog0411, 2018-12-19:
Thanks, removing gap fill definitely seemed to reduce my travel paths, I am showing almost no travel path now on the inside of object. However, now there's no fills in my perimeters so I don't think it will print good with no fill in between the perimeters to get nice solid cookie cutter outline.

Below photo show before and after with travel paths on gap fill and no gap fill

[image]

[image]

supermerill, 2018-12-20:
I don't have your stl to test, so it's a bit of guessing here.

Maybe you can increase the external perimeter width?

supermerill, 2019-02-02:
Maybe i will re-do the "don't cross perimeter" by adding a "follow external perimeter if you are on top of it, please". But not right now, sadly, i have other things to do before.

photog0411, 2020-01-01:
> 
> 
> Maybe i will re-do the "don't cross perimeter" by adding a "follow external perimeter if you are on top of it, please". But not right now, sadly, i have other things to do before.

Thank you :) No rush but it will help much

supermerill, 2020-01-09:
Note that Prusa has done some work on "don't cross perimeter" recently.

CCS86, 2021-06-16:
In this case, if you built a more precise model, with an even wall thickness, you could just size your widths to completely fill, with no need for gap fill. Then, there will be less need for extra travel.