supermerill/SuperSlicer #21: error in function Slic3r::ExtruderEntityCollection::as_polyline   when using support

photog0411, 2018-12-20:
### Version
1.41.2 Supermerill edition

### Operating system type + version
windows 10

### Behavior

I can't use any support, or I get this error. It won't let me export the gcode and it also doesn't let me preview the gcode.

I don't get this error in official slic3r PE, only your fork.

[image]
[image]

photog0411, 2018-12-20:
Happens with every .stl file

photog0411, 2018-12-20:
I'm assuming because it only happens in your fork, it must be one of your custom settings I have enabled elsewhere interfering with the support being generated successfully. I'm not sure which though. I disabled a bunch of them but still getting the error

supermerill, 2018-12-20:
I'm not at my home this week. I will see if I can patch it.
I modified a bit the support (to allow collection inside them). Maybe the recent changes broke something.

photog0411, 2018-12-21:
No rush. For now I use main slicer PE

supermerill, 2018-12-21:
cannot reproduce.
[image]

can you export the platter -> .3mf?

photog0411, 2018-12-21:
> 
> 
> cannot reproduce.
> [image]
> 
> [image]
> 
> can you export the platter -> .3mf?

No problem, I can't upload here so try this link to 3mf download:
https://transfernow.net/67bl01s81m5w It expires after 2 days.

Pardon the model I used for this, I know it can be printed without support by rotating it, but it is a support test model.

supermerill, 2018-12-21:
reproduced.

Problem is in the skirt. You can remove it to slice successfully.

photog0411, 2018-12-22:
i can confirm, removing the skirt does fix it .

Any idea why I can use skirt in official prusa fork, but not this fork? Seems like a bug.

supermerill, 2018-12-22:
It's complicated. Basically, i made some part of the codebase more generic (to handle ironing, i.e. two infills in one) and this function wasn't modified accordingly.
I will make a new release quickly.

photog0411, 2018-12-22:
Take your time, man, it's not a big deal now that we're know it can be
fixed by removing skirt.

closed 2019-01-18 by photog0411