supermerill/SuperSlicer #24: bug:  custom start gcode under printer not saving

photog0411, 2019-01-08:
### Version
1.42.0-alpha2

### Operating system type + version
win64

### Behavior
custom start gcode under printer not saving when profile is saved. It just reverts back to whatever the last input was when profile is reloaded.

photog0411, 2019-01-08:
Also, it appears ctrl+A (to select all text) is no longer working in the gcode text boxes. Possibly related.

photog0411, 2019-01-08:
> Corrected yesterday by prusa. Will be fixed in the next version.

ok thanks brother. I didn't know if this was also present in the prusa edition or just your fork -- I searched the prusa GitHub issue list but didn't see it posted, maybe I was using the wrong terms.

Patrik-svobodik, 2019-01-11:
@supermerill it looks like you almost abandoned this project, I hope it is not true but I have not seen any progress or merging of the master

supermerill, 2019-01-11:
@Patrik-svobodik ?
https://github.com/supermerill/Slic3r/releases

Patrik-svobodik, 2019-01-11:
@supermerill thanks, I see your progress on Zaribo-firmware, why dont you merge them together? Or are they already merged?

Patrik-svobodik, 2019-01-11:
Sorry, I mixed up two repos in my head

photog0411, 2019-01-18:
Thanks, it's prusa's error and they corrected it.

Brainlazy, 2019-02-01:
Although fixed in "Brim Ears edition" the recent pre-release, "Support edition", seems to manifest this problem again.

supermerill, 2019-02-02:
for the time being, you can ctrl+maj+a

closed 2019-02-04 by supermerill