supermerill/SuperSlicer #25: need confirmation - alpha3 shows up as alpha2?

photog0411, 2019-01-12:
### Version
1.42.0 alpha 3 - I THINK!

### Operating system type + version
win64

### Behavior

I installed the latest 1.42.0 ALPHA 3  release but it's showing up in the program as alpha 2 still? I just want confirmation if you are getting this too, or if maybe the release was compiled wrong.

[image]

photog0411, 2019-01-12:
i downloaded the official slic3r PE alpha 3, and it shows up correctly as the right alpha3 version, so it's not a main fork problem

supermerill, 2019-01-13:
Seems like prusa forgot to commit/push the file with the new version number. I forgot to verify.

photog0411, 2019-01-13:
> 
> 
> Seems like prusa forgot to commit/push the file with the new version number. I forgot to verify.

ok, so it's v3 but only says v2?

supermerill, 2019-01-13:
yes

photog0411, 2019-01-18:
Ok perfect :)

closed 2019-01-18 by photog0411