supermerill/SuperSlicer #26: [Feature Request] HTPLA as filament type, or allow user to set their own custom filament types

photog0411, 2019-01-18:
I use HTPLA as filament often, which requires higher temperature than normal PLA. But Slic3r don't support this filament right now. 

[image]

Reason for having this as an option would be so that gcode output naming when using placeholders can be differentiated from regular PLA...

[image]

I asking you to put it in your fork, rather than in main slic3r, because prusa never listen to any of my new feature ideas and they just close the topic.

supermerill, 2019-01-25:
It's a bug, should be corrected in a future release.

supermerill, 2019-02-04:
I tried to correct that but it 's too complicated. It's a gui stuff, and prusa may repair it themself. For now, I'll forfeit and wait to see what prusa will do for that.

supermerill, 2019-02-14:
I will add some options for the next release.

supermerill, 2019-02-14:
it's not released yet, but here the future format you should be able to write with the rubber-band:

`[input_filename_base]_20_{if filament_type[0]=~/other0/}HTPLA{elsif filament_type[0]=~/other1/}PLA+{else}{filament_type[0]}{endif}.gcode`

I added other0->other9 to make some room. I hope prusa will be able to make the field editable in the future.

note: the FLEX / SCAFF / PVA make the extrusion speed of the wiping tower slower, so i added also a new field below to be able to control this setting manually.

closed 2019-02-21 by supermerill