supermerill/SuperSlicer #27: Tree supports

Patrik-svobodik, 2019-01-18:
Just tree supports as cura or meshmixer has, they are already there for the SL1, so why not for FDM

supermerill, 2019-01-30:
prusa is working on that (for the sl1 support). So it should come in some weeks or months.

photog0411, 2019-02-02:
> 
> 
> prusa is working on that (for the sl1 support). So it should come in some weeks or months.

for just SL1 or for FDM too?

Patrik-svobodik, 2019-02-02:
Yes, because the functionality is already there for the SL1, just not for the FDM

MisterPrintf, 2019-04-15:
I'd definitely like to see this.

vertigo235, 2019-12-31:
ohhh I want this :D

zbrozek, 2020-07-24:
I just read [this](https://all3dp.com/2/cura-tree-support/) and now I covet them too.

supermerill, 2020-07-24:
For what i understand, you can see that the cura tree have thick branches and the current prusa sla one are thin, so it needs significant rewrite to make them suitable to fff.

Skaifer, 2020-10-08:
I hate Cura for its UI and overall its bulky. Right now I'm having zits on my prints, that do not present on any other slicer. But I love tree supports, that **only** Cura offers. Much less print time, easier to remove. Sad, that it's hard to implement.

forerunnert, 2020-11-05:
No need for significant rewrites, it's fine as it is!
When I was young I'd use to print 3D scans of people on commission and used Meshmixer SLA supports extensively for my FDM prints. I'd just dimension them to 3mm diameter and a 1mm tip.
No auto-generation either: the trick is to identify the spot where extrusion in the air happens first and add a support to that.

Currently I do the same with SuperSlicer: I have a SL1 profile, import my STL and manually add supports to that. Then export the STL including supports, open in FDM profile and scale up to achieve ~3mm diameter supports.
Similar to [this](https://blog.prusaprinters.org/wp-content/uploads/2018/04/tree_support.jpg)

hifihedgehog, 2022-01-06:
> > given you said that 2 years ago, do you have any more insight on their progress by now? kinda looks like they're not doing much for that feature :/
> 
> They tested it for fff but the result was worst than the actual support stuff. The PS 2.4 has some improvements. For tree support... maybe after.

Happy New Year! Now that PrusaSlicer 2.4 has been released and will be merged in the next version of SuperSlicer, where do we stand on this essential feature? For my high detail model prints, having working tree supports is the difference between me using SuperSlicer exclusively and me still having to use Cura to avoid failed prints.

supermerill, 2022-01-07:
I'll add it to the list of "big things todo", so it can be properly planned

mrkylegp, 2022-01-19:
I'd also love to see Tree Supports added so I don't have to use Cura sometimes.

blockfeed, 2022-03-23:
I'd also like to see this added. I'm really enjoying Superslicer and would appreciate not having to add these separately. Keep up the great work!

cron410, 2022-03-23:
> add these separately.

@blockfeed how are you adding them separately?

supermerill, 2022-03-24:
One method is to add them in an sla printer and then export the platter as stl.

nonchip, 2022-05-17:
@Lecso11 @tommylux @raimonds1503 @hifihedgehog how about yall stop spamming and just use the react function if you really want your + seen? -_-

theschles, 2022-10-20:
> One method is to add them in an sla printer and then export the platter as stl.

Hi @supermerill I tried this; the only options I saw were:
* Zig-Zag
* Cross
* Dynamic

Where's the equivalent option for "tree"?

Roycinger, 2022-12-17:
Any update on this?

tommylux, 2022-12-17:
> Any update on this?

in general? Tree supports and Organic supports are coming out in 2.6, check out youtube videos.

Tom

hifihedgehog, 2023-01-27:
> @Lecso11 @tommylux @raimonds1503 @hifihedgehog how about yall stop spamming and just use the react function if you really want your + seen? -_-

Boys, looks like we got a honeybadger out on the floor here with some ruffled feathers it seems. Take a chill pill and take a seat. If a post hasn't been made in months and I'm posting semi-annuallly, it isn't spamming so settle down there, sonny. The world isn't going to end from a ping in your queue every few months... or less. In fact, I know of many developers with more stable control of their emotions than yourself who appreciate gentle occasional reminders of this sort.

tommylux, 2023-01-27:
I was a little taken back I must admit, I was just promoting good enthusiasm and collaboration. But the feature is coming out in 2.6 and the new release looks really cool with organic supports,  embossed texts and more cut options. Waiting patiently for this massive release and keeping my eyes out. Kudos to the developers for making such a fantastic tool that brings joy and excitement to the people who use it.

nonchip, 2023-01-28:
soo if @hifihedgehog is done insulting and defaming me on multiple occasions for my single reminder above (by the way those "semi-annually" were 4 people posting the same "+1" in the span of 2 weeks, and funny how me reminding you of the fact that's uselessly pinging everyone involved is "emotionally unstable trolling" while your actively filling a topic with objectively useless messages is somehow "promoting the greater good"), can we get back on topic?

AuntJemima, 2023-03-01:
+1

This feature works exceptionally well in PrusaSlicer, and would be a strong improvement to SuperSlicer.
Another recent request of such: #3057

enzoescipy, 2023-04-24:
+1 however I know that supporting this would be such a horrible thing for developers. cheer up for all users and developing team !

mutlubasdas, 2024-01-03:
+1

I love SS and tree supports would be an amazing addition. I don't think I ever would need to use PS again :)

I really hope to see this update soon!

supermerill, 2024-01-16:
will be in 2.7, soon©

TonyRouse, 2024-01-29:
Wait, Issue #27 will be in v2.7? That's divine.

enzoescipy, 2024-02-04:
> will be in 2.7, soon©

oh my god supermerill, you are the god of us HOLY MOLY

reneas, 2024-12-24:
just cautiously asking if this is going to come anytime soon...? would be superb!