supermerill/SuperSlicer #28: Unable to build

Patrik-svobodik, 2019-01-26:
Hi, I am unable to build the masterPE branch, I am following the Prusa manual, I am able to build the original repo just fine, this is the error:

make[2]: *** [src/libslic3r/CMakeFiles/libslic3r.dir/pchheader.cpp.o] Error 1
make[1]: *** [src/libslic3r/CMakeFiles/libslic3r.dir/all] Error 2
make: *** [all] Error 2

supermerill, 2019-01-26:
use the commit 29d2d80202e017315cbeb1c4a671f5863955db5d

i tried to merge but i didn't have the time to build. I tried then to set the label on the "good" commit, but no luck with only the web interface

supermerill, 2019-01-26:
visual studio 2013?

Patrik-svobodik, 2019-01-28:
@supermerill did you look into it?

supermerill, 2019-01-28:
I just finished building the materPE:head. I has to add some std: in some places. I didn't had the time to push it before taking my plane, sorry.

I didn't encounter the " 'string.h' file not found". Maybe it's a problem on your system: your make didn't find it.

Patrik-svobodik, 2019-02-02:
Hi, when will you push it out?

supermerill, 2019-02-02:
now ;-p

Patrik-svobodik, 2019-02-21:
I am still unable to build it, even with the newest commits, there is
something weird

uniacke1, 2019-06-11:
Trying to build on Linux here with Boost 1.66, made a few tweaks for my system but fails at linking step, I get the error undefined reference to `boost::filesystem::path_traits::convert(wchar_t const*, wchar_t const*, std::string&, std::codecvt<wchar_t, char, __mbstate_t> const&)'

both Boost:fileysytem and boost:system libraries and devel packages installed.

uniacke1, 2019-06-12:
Update - solved. My issue was caused by differing ABIs on the compiler and my system's boost libraries from being built with different versions of gcc.

supermerill, 2019-06-27:
I have much less difficulties setting up the dev toolchain for compiling for 2013 & 2017 now. I will close that issue. Open a new one if a similar problem arise again.

closed 2019-06-27 by supermerill