supermerill/SuperSlicer #29: is this floor wall separation a possible setting m

photog0411, 2019-01-31:
Can't link the thread to the private Facebook group about this issue so I'll just post a photo of the discussion

[image]

photog0411, 2019-01-31:
More info on the proposed setting

[image]

supermerill, 2019-01-31:
Use thin wall with 100% threshold.

supermerill, 2019-06-27:
not much activities.

If you have a model & settings that create this issue, you can comment here again and link them, so i can try to test & investigate a bit on that.

In the meantime, i will close it as there are no feedback since February.

closed 2019-06-27 by supermerill

Photogad, 2020-02-01:
@supermerill 
> 
> Keep thin wall activated with no more than 100% threshold. (you can lower the threshold if you want, it only affect external bits, but 100% is "safer" if you don't need to print features smaller than nozzle size).
> 
> Also my gap fill algo is more reliable and my solid infill should also be more reliable (with the right setting).
> 
> edit: if the issue is the "big solid layer", in my fork you can change the size of the anchor area, in the infill tab. 3mm is the slic3r deault, 0mm is cura and my default is 1.5mm for "top" (default) and 2mm for bridge. I don"t advise to go below your nozzle size for the bridge anchor. (anchor solid infill by Xmm)
> 
> If it resolve your issues, feel free to post a link to my fork on facebook, i don't use facebook myself.

Sorry for delay. I did finally get to testing the settings above that you suggested, and it maybe helped a little bit, but the exterior lines still happen.

What I changed was anchor area to 1mm for top and 1.5mm for bridge. I don't know if I can safely make that much smaller.

Below is the actual print vs the 3d model comparison. The raised line on the outside of the print, is exactly where the inside "floor" of this "drawer" is. Really, the outside surface of this print should be flat.

[image]
[image]
[image]

Photogad, 2020-02-01:
I have now exactly created github about this issue entirely;
https://github.com/Photogad/Slic3r-Problem-66

But I do hope we can fix this using the custom settings in Slic3r++ since Prusa does not seem to want to fix it in their Prusa Slicer.

Exactly also here, we can see this issue mentioned on the Prusa Slicer git
https://github.com/prusa3d/PrusaSlicer/issues/3423
https://github.com/prusa3d/PrusaSlicer/issues/1719
https://github.com/prusa3d/PrusaSlicer/issues/991

supermerill, 2020-02-03:
If you have this issue with slic3r++, maybe i'ts because you have a "100% infill" at these layers and a "sparse infill" at the other, and the plastic retraction is then different.

prusaslicer has a very big "anchor" for bridge / top support, than can easily create "100% fill" layers.
By reducing "Anchor dsolid infill by X mm", you should be able to add some sparse infill and be more cura-like.

Photogad, 2020-02-03:
> If you have this issue with slic3r++, maybe i'ts because you have a "100% infill" at these layers and a "sparse infill" at the other, and the plastic retraction is then different.
>
> prusaslicer has a [...]

The issue is both in Prusa Slicer and Slic3r++, I'm assuming as always and referenced by the other issues, mostly an issue with how Slic3r program itself works and computes the slices.

I was able to reduce this problem quite a bit by using the below settings in Slic3r++:

[image]
[image]

However I'm not sure which of these 3 settings did actually help to fix the problem.

How far do you think I could safely lower the anchoring?? I guess I don't understand how it fully works. I suppose if I lower the anchoring too much, then my top surfaces would have gaps?

Photogad, 2020-02-03:
But here's what I don't understand.

The protruding line on the outside of the print does not even coincide where there would be solid infill touching the perimeter, so I don't understand how these settings fixed it.

For example below, I have drawn the blue line to show where along the outside surface of the print that the artifact (protruding line) happens,  however with the blue arrow I have pointed to where the solid infill is.

[image]

supermerill, 2020-02-03:
I'll do some test print this week-end.

Photogad, 2020-02-04:
@supermerill I do have a question about what you said below

> if the issue is the "big solid layer", in my fork you can change the size of the anchor area, in the infill tab. 3mm is the slic3r deault, 0mm is cura and my default is 1.5mm for "top" (default) and 2mm for bridge. 
> 

In slic3r++ for the description of the anchoring, it says % is based on the width of the perimeters.

Now it says the default is 150% for solid infill and 200% for the bridged infill anchoring.

Wouldn't 150% of perimeter width be 0.67mm (assuming 0.45mm perimeter on 0.40mm nozzle) and 200% would be 0.9mm?

So how does the slic3r default be 3mm?? Wouldn't slic3r default be 0.67mm?

Sorry to be annoying but to me this does not make sense.

My end goal here is having this anchoring setting as low as possible, but I would definitely like to use a % and not a fixed number, so that I could use this slic3r profile with multiple nozzle sizes

supermerill, 2020-02-04:
Yes, you can use the % that way.
Important note: 
* solid anchoring % is a % of the "shell" (total width from all perimeters), because with 100%, you can cover all perimeters.
* bridge anchoring M is a % of the external perimeter width, so with 100%, you only anchor into one perimeter.

prusa slicer default to 3mm because slic3r default to 3mm since age. I think the rational is to be on the safe side, in case there are a very low infill %. In my fork, this problem is reduced by the "denser infill" layer, even if it's not perfect.

chrzr, 2020-04-06:
> The issue is both in Prusa Slicer and Slic3r++, I'm assuming as always and referenced by the other issues, mostly an issue with how Slic3r program itself works and computes the slices.
>
> I was able [...]

This also solved my problems with the bulge occuring at the first top solid infill layer

Voidx, 2021-11-18:
this problem continue.

supermerill, 2021-11-18:
> this problem continue.

and it's still impossible to reproduce.
The only thing we found is when the layer time change drastically (so linked to a filament cooling issue)