supermerill/SuperSlicer #30: Feature request: Move purging volumes to Filaments Tab

3d-gussner, 2019-02-02:
### Version
Slic3r++ 1.42.0-alpha3.11

### Operating system type + version
Windows 10 (build 17134), 64-bit edition

### Feature Request
I opened this Feature Request at Prusa Slic3r github and they are looking for volunteers. As i am not a programmer and love your improvements to Slic3rPE I was thinking maybe you will look into it if that makes sense or not and how difficult it is to implement.
Here the link to the original Feature request and feedback i got https://github.com/prusa3d/Slic3r/issues/1282

BTW: Thanks for the great work.
My favorites at this moment (have to look for more awesome upgrades/features) are:
- Love the "Brim ears", i designed in the past my own for some potential lifting parts + ABS prints, but now no need for that anymore.
-The Octoprint cancel-object output option you implemented.

supermerill, 2019-02-02:
The goal is to replace the "minimal purge on wipe tower" by a "pigment density" setting and letting slic3r using it to determine the purge volume?

I will look at it, i also need something like that for my mmu prints. But I never looked at the purge code, so i don't know how hard it's to tune it.

3d-gussner, 2019-02-02:
Thanks. The goal is to get as less color bleeding as possible. And as it depends on the filament you use I was thinking to move it to filaments settings. It would be great if we could minimize the purge during changes to have less waste.

supermerill, 2019-02-04:
I implemented a thing in the last release. Please do some experiments on this to see if we can choose a good algo (and remove the bad ones), and if we can put some good default number (for e3Dv6 hotend, at least)

closed 2019-02-21 by supermerill