supermerill/SuperSlicer #32: dragging stl onto slic3r icon no longer works

photog0411, 2019-02-03:
### Version
_Version of Slic3r used goes here_
Latest release... Support edition

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
Win64
### Behavior

Can no longer drag stl file onto slic3r icon and have it get opened automatically. Have to now open slic3r first and drag stl file into slic3r window.

Last version before brim ears edition was working fine.

3d-gussner, 2019-02-03:
Can confirm that it is not working in Slic3r++ 1.42.0-alpha3.11 on Windows 10 (build 17134), 64-bit edition

photog0411, 2019-02-04:
working good in the newest release.

closed 2019-02-04 by photog0411