supermerill/SuperSlicer #34: Feature request: MM rainbow filament, zebra effect option and others

3d-gussner, 2019-02-03:
### Version
Slic3r++ 1.42.0-alpha3.11

### Operating system type + version
Windows 10 (build 17134), 64-bit edition

### Feature Request
Hi,

maybe some other MMU functions would be interesting, like the Palette 2 has:
- Gradient Mode where there is a continuous material change without purge block (maybe wipe oozing filament inside the object) https://support.mosaicmfg.com/hc/en-us/articles/360011230493-Gradient-Mode-Palette-2-
- Pattern Mode https://support.mosaicmfg.com/hc/en-us/articles/360011133574-Pattern-Mode-Palette-2-
- Random Mode https://support.mosaicmfg.com/hc/en-us/articles/360011132614-Random-Mode-Palette-2-

I really like these modes as you don't have to have multi material STL files. You just have aone STL part you want to print with bit more shine to it.

I opened a feature request on the Prusa Github, but as they are busy with lot of other things it might take some time https://github.com/prusa3d/Slic3r/issues/1283

Don't know if that is interesting for you to have a look into it.

supermerill, 2019-02-21:
I understand the last two:

* Pattern Mode : you enter an array of extruder id (one entry per layer) and this is the pattern. (1 or 0 filament change per layer)

* Random Mode: at each layer, select a random extruder and if it's different, change the filament.

But the gradient mode is a bit more cryptic.
Does it purge or not?

What i understand:
* at the begin of the print, almost all layers are of color0
* at the middle, half layers are  color0 andfd the other half are color1 (with a filament change at each layer)
* at the end of the print, almost all layers are of color1

Is it that?

3d-gussner, 2019-03-06:
Hi, sorry for the late answer.
Did you watch the video https://youtu.be/eBwBqqdqvCQ it describes it quite good. Watch out for the mm and cm and m in the different menu.
https://youtu.be/eBwBqqdqvCQ?t=164 you can see the result.

1. In the beginning the print will be in color0 until max height
2. the next part will be color1 min height
3. next part will be x% lower than the max height in color0
4. next part will be x% higher than the min height in color1
5. repeat 3 and 4 until the color0 has reached min height and color1 reached max height