supermerill/SuperSlicer #35: Weird gaps in perimeters of all models, but especially benchy

Murloc992, 2019-02-03:
### Version
1.41.2 

### Operating system type + version
Windows 10 64bit

### Behavior
When printing the lower part (~6mm) of a benchy with Slic3r++ I am getting some interesting gaps.

The gaps are not appearing when using Slic3r PE 1.41.2 with identical .ini files as they use the same settings folder.

This is a comparison picture:
[image]

SLic3r++ is on the right.

To reproduce it, just print the included small part of 3D benchy, it's a small print and a short one too.

I am wondering, maybe it's a settings issue? Yet disabling small gap detection and enabling thin walls did nothing.

I now have a ton of benchy bottoms trying to witch hunt this. :)

#### STL/Config (.ZIP) where problem occurs
[bugreport.zip](https://github.com/supermerill/Slic3r/files/2825478/bugreport.zip)

supermerill, 2019-02-04:
Can you test with tjhe seam to "hidden" (it's the default setting for slic3rPE, i just renamed it)?

also, can you send me your two gcode files? (the one with the good benchy, and the one with the bad benchy)

Murloc992, 2019-02-06:
I will do it when I find the time, meanwhile another issue has risen, I will make a report soon..

Murloc992, 2019-02-06:
I have added a comment on the vase mode issue: https://github.com/supermerill/Slic3r/issues/5

Murloc992, 2019-03-02:
Sorry for not giving an update for so long.. I will try to squeeze some prints inbetween huge orders I have today..

supermerill, 2019-11-06:
I close this for inactivity.
If someone else find a degradation in a print vs prusaslicer, plaes re-open this issue and post the two project file (.3mf in a zip, from prusaslicer & slic3r++)

closed 2019-11-06 by supermerill