supermerill/SuperSlicer #37: Ironing: whispy bits of filament....

LaserBee, 2019-02-09:
### Version
Slic3r++_1.42.0-alpha4.0_190204

### Operating system type + version
Win 10
### Behavior

Just tried ironing for the first time, and I am getting odd whispy bits of filament deposited on top of the print. The fan is on 100%, so I am wondering if it's blowing some off the extruder? It's a really odd problem, which has never happened before without ironing...

[image]

supermerill, 2019-02-09:
yeah, ironing is buggy on the last version. Get the previous one or wait the next one (2-3 days).

LaserBee, 2019-02-09:
Thanks for the reply. While I've got you here, can I make a feature request?

The top layer of flat objects is always drawn diagonally, from the lower
left to the top right. Would it be possible to make it draw the top layer
horizontally, so the lines run left to right instead?

supermerill, 2019-02-09:
for ironing?

for rectilinear, you can use the Angle -> fill setting

LaserBee, 2019-02-09:
Ah, I hadn't spotted that! That will be a big help, thanks!

closed 2019-02-13 by supermerill