supermerill/SuperSlicer #38: Feature Request: Iron all layers and clear print options

nubnubbud, 2019-02-15:
1.42.0-alpha5.0+unknown
Windows 10 home version 1803

This is a feature request/continuation of the studies and changes logged on Reddit,following these successful clear prints obtained from the cura slicer: 
[image]

essentially, due to the changes made after the feature request seen on reddit, Ironed infill has been added successfully to the dev build. It works as intended in code, _but_ does not currently produce the same results in other slicers(cura) and is therefore unfinished, for the time being, The main source of scattering within the model is the presence of microscopic bubbles, both in lines pressed between the extrusions, and caught under the extrusions due to printing on a roughly ironed surface. Here are some solid prints with clear filament. both of their top layers have been sanded to a polish for comparison:

[image]

without bubbles (cura code hand-modified with ironing every layer):
[image]

with bubbles (Slic3r code direct from program)
[image]

I'm not sure of the exact cause of the bubbles in this case, but the common solutions are: extruding more, extruding 10% as much as normal, just for ironing, ironing in smaller passes, ironing a little hotter, and selectively cooling either the ironing, first extrusion, both, or none.

with all these options, it may be prudent to add ironing as its own, advanced section in Layer Options

supermerill, 2019-02-15:
> I'm not sure of the exact cause of the bubbles in this case, but they appear to be a combination of rough ironing and gaps between extrusions

Two way to improve:
 * you can compile slic3r and you can modify the settings for ironing (in fillsmooth.hpp, i can explain the values)
 * You can't and i have to buy some translucent petg and spend time to debug the feature (will not happen before months)

>with all these possible options, it may be prudent to add ironing as its own, advanced section in Layer Options, instead of an infill type, seeing is ironing should also apply to the shell.

if you don't want any perimeter, just set the number to 0, unless i misunderstand.

nubnubbud, 2019-02-15:
> > I'm not sure of the exact cause of the bubbles in this case, but they appear to be a combination of rough ironing and gaps between extrusions
> 
> Two way to improve:
> 
> * you can compile slic3r and you can modify the settings for ironing (in fillsmooth.hpp, i can explain the values)
I'll gladly do the tests, but I won't have access to my printer for a week. if you want to test, though, Any clear PETG will work. I'm using $17 stronghero PETG because it's one of the hardest to make clear. Once we have the correct settings, almost any filament should become clear. I'll gladly work on it after next week. I might as well study how git works and brush up on my c++ while I'm away too.

> > with all these possible options, it may be prudent to add ironing as its own, advanced section in Layer Options, instead of an infill type, seeing is ironing should also apply to the shell.
> 
> if you don't want any perimeter, just set the number to 0, unless i misunderstand.
I mean to add an ironing section alongside the other advanced options like flow and overlap. Cura has: "ironing pattern", "ironing line spacing", "ironing flow", "ironing inset", "ironing speed", "ironing acceleration", and "ironing jerk"
all in the "shell" section, or what would be slic3r's "perimeter"section.

I don't think we need all of that, as welcome as it would be. acceleration and jerk aren't needed, but the others like flow, line spacing, and inset are likely the key to clear prints. I know cura defaults to 10% flow, and 20mm's, with .1mm line spacing.

nubnubbud, 2019-02-15:
By increasing extrusion width and decreasing infill speed to 20mm/s I was able to greatly eliminate the inconsistencies between layers, and remove most bubbles except between extrusions, resilting in more light let through. those bubbles between extrusions actually got bigger for some reason. That's all I can do  for now.
[image]

supermerill, 2019-12-12:
Some new settings have been added, you can try them.
Reopen if you have some new input to share.

nubnubbud, 2019-12-20:
Alright! I've gotten my way around this, tested a few dozen times, took the resulting gcode, and spliced it with custom gcode to test my theories, and I believe I've come up with the final things that need to be done to complete this feature in a way that will set slic3r apart from the competition for a bit~... in any case here are my new findings! This is my first print, the classic XYZ test cube, with all standard settings for PETG and ironing infill.
[image]
I noticed immediately that it was clear! ironing is definitely working, though you need to turn off the fan for the entire print. it may only need to be off for part of it, but I played it safe. The issue is that the edges, the perimeter, is not clear, but the insides are. I decided, then, that I would test with only the insides, and turned perimeters to 0:
[image]
[image]

(ignore the tabs, they were to test if bubbles could be pushed around. they can't really.) but it was a great success, the light shone through, and the ridges from the rectilinear infill pattern acted as a fresnel lens, meaning the insides were somewhere around optically clear! I wasn't satisfied though- those ridges would be troublesome for both mechanical or aesthetic prints, I sanded it down to chack its clarity, and went to work splicing together some custom gcode.
First off, the final issue here is very visible:
[image]
[image]
the perimiter walls are trapping bubbles, making them scatter light, whereas my custom code produced much cleaner results
At the end of that, I was left with this: 
[image]
this part had no post processing, and there was definitely some small dimensional errors, just under half a millimeter, from the excess plastic getting spread to a couple sides preferentially. 

My custom Gcode was made by taking a sample of the ironing code in ironing mode, and putting that at the "before next layer" code input box, setting the printer to relative steps at the beginning and reversing that at the end of the custom gcode. 

I realized that because my ironing gcode was always going to be the same direction, rectilinear infill would not work. Luckily:
**because my ironing code was separate from the infill code, I was free to choose concentric infill**
**I was free to adjust how close the ironing got to the edge for the same reason**
**the concentric infill went from inside to out, removing the line of bubbles that a travel move would make**

so, there are several options available to achieve this effect:
1. make ironing a separate function or process from the infill and line types, a kind of post-processing applied to every layer, and add an option to widen and shrink the ironed areas
2. add an option to the ironing settings, to expand the ironing area into the perimeters, or by a set distance
3. add a check bod which will include perimeter walls as iron-able

of the options that I found, number one is the most effective, but also the most invasive, likely, seeing as, as far as I can tell, ironing is currently tethered to the infill, and making it a separate process could be a lot of work, but it would also make it an easier function to use, enough that it might even make it into the advanced features as an uncommon, but useful and not terribly complex tool.