supermerill/SuperSlicer #39: Crashing on slicing

LaserBee, 2019-02-15:
### Version
Slic3r++_1.42.0-alpha5.1_190214

### Operating system type + version
Windows 10

### Behavior
Trying to slice this file. Program freezes at about 20% of progress bar. 

[STL files.zip](https://github.com/supermerill/Slic3r/files/2870418/STL.files.zip)

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

supermerill, 2019-02-15:
Can't reproduce. It's just a bit slow.
You can remove "extra perimeter, it's not useful with no overhangs and it's slow.

But you rmodel show an artefact with the "only one perimeter on top", i will correct that.

LaserBee, 2019-02-16:
You're right, I was being impatient. If I start slicing, and then click on
the screen anywhere (INCLUDING the cancel button), it crashes. If I leave
it alone, it works fine!

I do find it still less stable than the "official" PE release, but the huge
number of fantastic features more than makes up for it. The "one perimiter
on top" is a stroke of absolute genius, it makes my work look *SO MUCH*
better.

Thanks for your hard work!

supermerill, 2019-02-16:
> If I start slicing, and then click on the screen anywhere (INCLUDING the cancel button), it crashes.

It doesn't crash if you click on "wait". I'll see if i can add some pauses to let the ui refresh more often.

closed 2019-03-14 by supermerill