supermerill/SuperSlicer #40: Cannot set a cut height

Sidic101, 2019-02-17:
### Version
"boring++"

### Operating system type + version
Windows 10 10.0.17763 build 17763 according to System information

### Behavior
When I click the "Cut" button, I can't set the cut height. Pressing numbers on the numpad rotates the view and does not insert the numbers into the textbox, the number bar numbers do nothing. Pressing delete or trying to use ctrl+v also only affects the scene, not the input texbox.

Sidic101, 2019-02-17:
I can still use the orange box thingy and change the cut height roughly with the mouse.

Sidic101, 2019-02-17:
The issue comes from the current build of PE and hasn't yet been fixed https://github.com/prusa3d/Slic3r/issues/1572

Sidic101, 2019-03-02:
It appears Prusa has fixed the issue, alpha7 is supposed to have it working.

Sidic101, 2019-03-02:
Prusa has fixed the issue in alpha7.

supermerill, 2019-03-14:
should be fixed in my latest release, so.

closed 2019-03-14 by supermerill