supermerill/SuperSlicer #41: Feature Request: 3DConnextion Space Mouse support

LaserBee, 2019-02-17:
### Operating system type + version
Windows 10

_Is this a new feature request?_

Yes

Many people who do 3D design use a 3DConnextion Space Mouse. This tool works brilliant in things like Fusion360 and Blender, for navigating around a model, zooming, panning, &c. 

It would be helpful to have it work in Slic3r, too. It's not a major thing, but it would mean my work-flow between the design and the slicing would be improved!

Thanks for the consideration.

supermerill, 2019-02-18:
It's for the camera mouvement (rotation/translation)?

LaserBee, 2019-02-18:
Yes, exactly. Rotation, panning, zooming. It's not a mouse, really, you use
it *with* a mouse in the other hand.

supermerill, 2019-02-18:
Most of the gui stuff is work-in-progress by Prusa research, so it may change quickly month after month.
Also, i don't want to tinker it because it's really boring and error-prone coding.
Also i don't have this sort of mouse.

Sorry, but this has a really really low priority for me, (taking weeks of free time for something I can't use...)
I can accept a pull request, but i can't do the code.

supermerill, 2019-11-25:
seems like prusa is working on that.
So you'll be able to use your mouse with the next version of prusaslicer / when i'll merge it here.

closed 2020-01-10 by supermerill