supermerill/SuperSlicer #42: Would lilke to see Adaptive Slicing / Quality added

OopAck, 2019-02-22:
### Version
_Version of Slic3r used goes here_

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

guestisp, 2019-04-26:
+100 for this :)

supermerill, 2019-04-26:
Not a priority for now. 
I want to do a complete rework of the layer height assignment algo before.

supermerill, 2019-12-12:
As the code was here, i re-activated it.
Also, prusa is working on something for the next version, so the old algorithm isn't going to stay (i try it a bit, imo it's shitty).

closed 2020-01-10 by supermerill