supermerill/SuperSlicer #43: Gap in external wall when using some types of seam alignment

ewb-git, 2019-02-23:
[image]

Noticed some really strange edge behavior on my first try of this slic3r branch. It looks like these gaps/harsh edges in the perimeter line.

After playing a little bit, seems the issue is due to the seam alignment. For example, random or rear creates a smooth seem in the picture, whereas nearest/aligned create this very ugly edge.

supermerill, 2019-02-23:
It's working as intended. that's the point where it cut the loop to move in/out. If you see the retraction point not at the same place, it's because of "wipe on retraction".

> seam alignment

The prusa default is "nearest", but it was not very "near". I changed it to "hidden" and created a new "nearest" that really prioritize the distance.

tl/dr: The default behavior should be "hidden", check if it's the case.

ewb-git, 2019-03-02:
Hmm...I’ve loaded another model and compared against a stock version of slic3r. Even using “hidden”, the quality of the seam, seam shape, seam position, etc looks much worse. There are a lot more seams in the “hidden” that stock. The corners of the seams have an irregular shape also...

Is there a version of this with stock seams and ironing enabled?

supermerill, 2019-03-02:
Strange, the "hidden" should be exactly the same as the prusa "nearest". I will look at it.

supermerill, 2019-03-03:
[image]

see an exemple of comparison about slic3rPE / slic3r++
apart a diffeence in the direction of the infill & the skirt (something that shouldn't do any harm), the seam is exactly the same.

ewb-git, 2019-03-03:
I will get the remaining files shortly.

Here I'm attaching photos of stock aligned, vs Supermerill "hidden. There is a big difference...
[image]
[image]
[image]

ewb-git, 2019-03-03:
The file was too big to attach, so here is an upload link to the slic3r folder zipped:
https://www.amazon.com/clouddrive/share/1B4Ym7mIi4uZTkHBUW1eBSWZtl0RIFXk4G4MOsSKUb2

For this, I don't think I need to upload any stls. It seems to occur with any stl that has seams in it.

supermerill, 2019-03-03:
You linked my software, that's not useful at all..

What you need to do:
* load your model and settings
* slice it and see that's wrong
* menu "File" -> "save as" and selct "my documents" and enter a good name like "my problem with seams"
* go into the file explorer at "my documents"
* right-clic on the "my problem with seams.3mf" -> compress/zip
* drag it here (or if your model is really enormous, into your cloud drive)

ewb-git, 2019-03-03:
Ok I can do this, sorry I guess I misunderstood what you meant by a "project file". Unfortunately for the setup I didn't save the config/project before closing.

I'm currently trying to get my bear upgrade printed out right now...so let me try and get back to this sometimes later in the week or early next week. I'll make a formal test, with better test parameters and more control and save the project files and upload them.

It's possible I did something wrong in the config, just I don't have time to do more tests at this moment since I'm in bear printing mode :p

supermerill, 2019-03-03:
ok, no problem, i can wait.

supermerill, 2019-12-12:
The issue is always here?

supermerill, 2020-07-03:
closing it as it's a too old issue, maybe already corrected.

closed 2020-07-03 by supermerill