supermerill/SuperSlicer #45: Wipe Tower error

gizzburn, 2019-02-24:
### Version
_Version of Slic3r used goes here_
1.42.0.Alpha 6
_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
Windows 10 Home 1803  17134.590
### Behavior
* _Describe the problem_
When I select to use the wipe tower in a dual print it pops this error up 

The Wipe Tower currently supports the non-soluble supports only if they are printed with the current extruder without triggering a tool change. (both support_material_extruder and support_material_interface_extruder need to be set to 0).

The above are already set to 0

* _Steps needed to reproduce the problem_

Just select use wipe tower or try to slice it.

  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_

It wont let me go any further than this
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

supermerill, 2019-02-24:
I can't reproduce.
Plase dsave your project, zip it and drag it here.

gizzburn, 2019-02-24:
See attached file.
[Slicr3_Error.zip](https://github.com/supermerill/Slic3r/files/2897887/Slicr3_Error.zip)

closed 2019-03-14 by supermerill