supermerill/SuperSlicer #47: linux compile problems

supermerill, 2019-02-26:
@mperino asked for direction for a linux build:

> Any tutorial for the Linux build? THe page for it is kinda empty :(
> https://github.com/supermerill/Slic3r/blob/master/doc/How%20to%20build%20-%20UNIX.md

me:

>I left the master branch on the last (december 2018) "stable" and used the masterPE for the new ui (and no perl, new build stack), as long as it's on alpha.

>you should checkout the masterPE branch and use this : >https://github.com/supermerill/Slic3r/blob/masterPE/doc/How%20to%20build%20-%20Linux%20et%20al.md

mperino :

>Thank you, That got me closer, but alas it puts me into a compile error: 
> ^~~~ 
> CMakeFiles/Makefile2:154: recipe for target

me:

> you forgot to copy the lines before the arrow. And you should open a "compile on linux" issue, that one is for macos.

mperino:
```

/home/mperino/3d/Software/supermerril-slicer/Slic3r-master/xs/src/libslic3r/TriangleMesh.hpp:112:110:
warning:   base ‘Slic3r::Point’ [-Wreorder]
/home/mperino/3d/Software/supermerril-slicer/Slic3r-master/xs/src/libslic3r/TriangleMesh.hpp:112:5:
warning:   when initialized here [-Wreorder]
     IntersectionPoint(const IntersectionReference &ir, const Point &pt) :
IntersectionReference(ir), Point(pt) {}
     ^~~~~~~~~~~~~~~~~
In file included from
/home/mperino/3d/Software/supermerril-slicer/Slic3r-master/xs/src/libslic3r/PerimeterGenerator.cpp:16:0:
/home/mperino/3d/Software/supermerril-slicer/Slic3r-master/xs/src/libslic3r/SVG.hpp:
In constructor ‘Slic3r::SVG::SVG(const char*)’:
/home/mperino/3d/Software/supermerril-slicer/Slic3r-master/xs/src/libslic3r/SVG.hpp:81:17:
warning: ‘Slic3r::SVG::filename’ will be initialized after [-Wreorder]
     std::string filename;
                 ^~~~~~~~
In file included from
/home/mperino/3d/Software/supermerril-slicer/Slic3r-master/xs/src/libslic3r/PerimeterGenerator.cpp:16:0:
/home/mperino/3d/Software/supermerril-slicer/Slic3r-master/xs/src/libslic3r/SVG.hpp:19:10:
warning:   ‘bool Slic3r::SVG::flipY’ [-Wreorder]
     bool flipY;
          ^~~~~
/home/mperino/3d/Software/supermerril-slicer/Slic3r-master/xs/src/libslic3r/SVG.hpp:21:5:
warning:   when initialized here [-Wreorder]
     SVG(const char* afilename) :
     ^~~
In file included from
/home/mperino/3d/Software/supermerril-slicer/Slic3r-master/xs/src/libslic3r/PerimeterGenerator.cpp:16:0:
/home/mperino/3d/Software/supermerril-slicer/Slic3r-master/xs/src/libslic3r/SVG.hpp:
In constructor ‘Slic3r::SVG::SVG(const char*, const Slic3r::BoundingBox&,
coord_t, bool)’:
/home/mperino/3d/Software/supermerril-slicer/Slic3r-master/xs/src/libslic3r/SVG.hpp:81:17:
warning: ‘Slic3r::SVG::filename’ will be initialized after [-Wreorder]
     std::string filename;
                 ^~~~~~~~
In file included from
/home/mperino/3d/Software/supermerril-slicer/Slic3r-master/xs/src/libslic3r/PerimeterGenerator.cpp:16:0:
/home/mperino/3d/Software/supermerril-slicer/Slic3r-master/xs/src/libslic3r/SVG.hpp:18:11:
warning:   ‘Slic3r::Point Slic3r::SVG::origin’ [-Wreorder]
     Point origin;
           ^~~~~~
/home/mperino/3d/Software/supermerril-slicer/Slic3r-master/xs/src/libslic3r/SVG.hpp:24:5:
warning:   when initialized here [-Wreorder]
     SVG(const char* afilename, const BoundingBox &bbox, const coord_t
bbox_offset = scale_(1.), bool aflipY = false) :
     ^~~
In file included from
[... 170 lines cut ...]
>]’:
/home/mperino/3d/Software/supermerril-slicer/Slic3r-master/xs/src/libslic3r/PlaceholderParser.cpp:484:106:
 required from ‘static void
Slic3r::client::expr<Iterator>::regex_matches(Slic3r::client::expr<Iterator>&,
boost::iterator_range<IteratorT>&) [with Iterator =
__gnu_cxx::__normal_iterator<const char*, std::__cxx11::basic_string<char>
>]’
/home/mperino/3d/Software/supermerril-slicer/Slic3r-master/xs/src/libslic3r/PlaceholderParser.cpp:1000:72:
 required from
‘Slic3r::client::macro_processor<Iterator>::macro_processor() [with
Iterator = __gnu_cxx::__normal_iterator<const char*,
std::__cxx11::basic_string<char> >]’
/home/mperino/3d/Software/supermerril-slicer/Slic3r-master/xs/src/libslic3r/PlaceholderParser.cpp:1180:33:
 required from here
/home/mperino/3d/Software/supermerril-slicer/Slic3r-master/xs/src/libslic3r/PlaceholderParser.cpp:462:32:
warning: unused variable ‘mask’ [-Wunused-variable]
             const std::string *mask     = nullptr;
                                ^~~~
CMakeFiles/Makefile2:154: recipe for target
'xs/CMakeFiles/libslic3r.dir/all' failed
make[1]: *** [xs/CMakeFiles/libslic3r.dir/all] Error 2
Makefile:162: recipe for target 'all' failed
make: *** [all] Error 2

```

supermerill, 2019-02-26:
@mperino : should be corrected on the last commit of the masterPE branch.

mperino, 2019-02-26:
I pulled the latest from your git.  Same error.  The latest from the prusa git ( "1.42.0-alpha5") compiles fine with the same process bellow..

Here's the process I used:
1. Install Prereq's:

 $ sudo apt-get install build-essential libgtk2.0-dev libwxgtk3.0-dev libwx-perl libmodule-build-perl git cpanminus libextutils-cppguess-perl libboost-all-dev libxmu-dev liblocal-lib-perl wx-common libopengl-perl libwx-glcanvas-perl libtbb-dev libxmu-dev freeglut3-dev libwxgtk-media3.0-dev libboost-thread-dev libboost-system-dev libboost-filesystem-dev libcurl4-openssl-dev libxs-dev libxs2
 

2. Uncompress or copy in the code from git:
$ unzip Master.zip

3. cd into the extracted files and create a build directory:
$ cd Slic3r-master
$ mkdir build
$ cd build

4. Run the cmake:
$ cmake ..

5. Compile (in the example bellow I use 3 cores -j3):
$ cmake -j3

Supermerill build (latest) gives:
In file included from /home/mperino/3d/Software/Meril-slicer/xs/src/libslic3r/PerimeterGenerator.cpp:1:0:
/home/mperino/3d/Software/Meril-slicer/xs/src/libslic3r/PerimeterGenerator.hpp: In constructor ‘Slic3r::PerimeterGeneratorLoop::PerimeterGeneratorLoop(Slic3r::Polygon, short unsigned int, bool)’:
/home/mperino/3d/Software/Meril-slicer/xs/src/libslic3r/PerimeterGenerator.hpp:34:20: warning: ‘Slic3r::PerimeterGeneratorLoop::depth’ will be initialized after [-Wreorder]
     unsigned short depth;
                    ^~~~~
/home/mperino/3d/Software/Meril-slicer/xs/src/libslic3r/PerimeterGenerator.hpp:32:10: warning:   ‘bool Slic3r::PerimeterGeneratorLoop::is_overhang’ [-Wreorder]
     bool is_overhang;
          ^~~~~~~~~~~
/home/mperino/3d/Software/Meril-slicer/xs/src/libslic3r/PerimeterGenerator.hpp:39:5: warning:   when initialized here [-Wreorder]
     PerimeterGeneratorLoop(Polygon polygon, unsigned short depth, bool is_contour) :
     ^~~~~~~~~~~~~~~~~~~~~~
/home/mperino/3d/Software/Meril-slicer/xs/src/libslic3r/PerimeterGenerator.hpp: In constructor ‘Slic3r::PerimeterGeneratorLoop::PerimeterGeneratorLoop(Slic3r::Polygon, short unsigned int, bool, bool)’:
/home/mperino/3d/Software/Meril-slicer/xs/src/libslic3r/PerimeterGenerator.hpp:34:20: warning: ‘Slic3r::PerimeterGeneratorLoop::depth’ will be initialized after [-Wreorder]
     unsigned short depth;
                    ^~~~~
/home/mperino/3d/Software/Meril-slicer/xs/src/libslic3r/PerimeterGenerator.hpp:32:10: warning:   ‘bool Slic3r::PerimeterGeneratorLoop::is_overhang’ [-Wreorder]
     bool is_overhang;
          ^~~~~~~~~~~
/home/mperino/3d/Software/Meril-slicer/xs/src/libslic3r/PerimeterGenerator.hpp:41:5: warning:   when initialized here [-Wreorder]
     PerimeterGeneratorLoop(Polygon polygon, unsigned short depth, bool is_contour, bool is_overhang) :
     ^~~~~~~~~~~~~~~~~~~~~~
/home/mperino/3d/Software/Meril-slicer/xs/src/libslic3r/PerimeterGenerator.hpp: In constructor ‘Slic3r::PerimeterGenerator::PerimeterGenerator(const Slic3r::SurfaceCollection*, double, Slic3r::Flow, Slic3r::PrintRegionConfig*, Slic3r::PrintObjectConfig*, Slic3r::PrintConfig*, Slic3r::ExtrusionEntityCollection*, Slic3r::ExtrusionEntityCollection*, Slic3r::SurfaceCollection*)’:
/home/mperino/3d/Software/Meril-slicer/xs/src/libslic3r/PerimeterGenerator.hpp:56:32: warning: ‘Slic3r::PerimeterGenerator::lower_slices’ will be initialized after [-Wreorder]
     const ExPolygonCollection* lower_slices;
                                ^~~~~~~~~~~~
/home/mperino/3d/Software/Meril-slicer/xs/src/libslic3r/PerimeterGenerator.hpp:55:32: warning:   ‘const Slic3r::ExPolygonCollection* Slic3r::PerimeterGenerator::upper_slices’ [-Wreorder]
     const ExPolygonCollection* upper_slices;
                                ^~~~~~~~~~~~
/home/mperino/3d/Software/Meril-slicer/xs/src/libslic3r/PerimeterGenerator.hpp:72:5: warning:   when initialized here [-Wreorder]
     PerimeterGenerator(
     ^~~~~~~~~~~~~~~~~~
In file included from /home/mperino/3d/Software/Meril-slicer/xs/src/libslic3r/Geometry.hpp:6:0,
                 from /home/mperino/3d/Software/Meril-slicer/xs/src/libslic3r/PerimeterGenerator.cpp:5:
/home/mperino/3d/Software/Meril-slicer/xs/src/libslic3r/MedialAxis.hpp: In constructor ‘Slic3r::MedialAxis::MedialAxis(const Slic3r::ExPolygon&, const Slic3r::ExPolygon&, double, double, double)’:
/home/mperino/3d/Software/Meril-slicer/xs/src/libslic3r/MedialAxis.hpp:23:26: warning: ‘Slic3r::MedialAxis::surface’ will be initialized after [-Wreorder]
         const ExPolygon& surface;
                          ^~~~~~~
/home/mperino/3d/Software/Meril-slicer/xs/src/libslic3r/MedialAxis.hpp:22:26: warning:   ‘const Slic3r::ExPolygon& Slic3r::MedialAxis::bounds’ [-Wreorder]
         const ExPolygon& bounds;
                          ^~~~~~
/home/mperino/3d/Software/Meril-slicer/xs/src/libslic3r/MedialAxis.hpp:28:9: warning:   when initialized here [-Wreorder]
         MedialAxis(const ExPolygon &_expolygon, const ExPolygon &_bounds, const double _max_width, const double _min_width, const double _height)
         ^~~~~~~~~~
In file included from /home/mperino/3d/Software/Meril-slicer/xs/src/libslic3r/SVG.hpp:8:0,
                 from /home/mperino/3d/Software/Meril-slicer/xs/src/libslic3r/PerimeterGenerator.cpp:16:
/home/mperino/3d/Software/Meril-slicer/xs/src/libslic3r/TriangleMesh.hpp: In constructor ‘Slic3r::IntersectionPoint::IntersectionPoint(int, int, const Slic3r::Point&)’:
[... cut ...]
         return true;
         ^~~~~~
xs/CMakeFiles/libslic3r.dir/build.make:1262: recipe for target 'xs/CMakeFiles/libslic3r.dir/src/libslic3r/PerimeterGenerator.cpp.o' failed
make[2]: *** [xs/CMakeFiles/libslic3r.dir/src/libslic3r/PerimeterGenerator.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:154: recipe for target 'xs/CMakeFiles/libslic3r.dir/all' failed
make[1]: *** [xs/CMakeFiles/libslic3r.dir/all] Error 2
Makefile:162: recipe for target 'all' failed
make: *** [all] Error 2

supermerill, 2019-02-26:
are you really sure?

that's what i have in "/home/mperino/3d/Software/Meril-slicer/xs/src/libslic3r/PerimeterGenerator.cpp:876" : 

`            }`

not

`                 coord_t dot = direction.dot(Line(direction_polyline.points.back(), direction_polyline.points.front()));`

and the last commit (see below) change the definition of dot from 

`coord_t dot(Line &l2)`

as seen in your log to 

`coord_t Line::dot(const Line &l2)`

supermerill, 2019-02-26:
There are a problem, you are not on the masterPE branch (there are now almost nothing in the xs folder).

do 'git checkout masterPE'

mperino, 2019-02-26:
I also tried both the latest "Stable" and "Try" source.tar.gz's.

I just Changed branches (to the proper masterPE) and have a different error:
from Branch: masterPE, "bugfix #47":

Scanning dependencies of target gtest_main
[ 53%] Building CXX object googletest-build/CMakeFiles/gtest_main.dir/src/gtest_main.cc.o
[ 54%] Linking CXX static library libgtest_main.a
[ 54%] Built target gtest_main
Scanning dependencies of target tests_clipper_nlopt
[ 54%] Building CXX object src/libnest2d/tests/CMakeFiles/tests_clipper_nlopt.dir/printer_parts.cpp.o
[ 54%] Building CXX object src/libnest2d/tests/CMakeFiles/tests_clipper_nlopt.dir/test.cpp.o
In file included from /home/mperino/3d/Software/Slic3r-masterPE/src/libslic3r/Polygon.hpp:7:0,
                 from /home/mperino/3d/Software/Slic3r-masterPE/src/libslic3r/BoundingBox.hpp:6,
                 from /home/mperino/3d/Software/Slic3r-masterPE/build/src/libslic3r/libslic3r_pch/pchheader.hpp:105:
/home/mperino/3d/Software/Slic3r-masterPE/src/libslic3r/Line.hpp:50:13: error: extra qualification ‘Slic3r::Line::’ on member ‘dot’ [-fpermissive]
     coord_t Line::dot(const Line &l2) const { return vector().dot(l2.vector()); }
             ^~~~
In file included from /home/mperino/3d/Software/Slic3r-masterPE/src/libnest2d/tests/test.cpp:1:0:
/home/mperino/3d/Software/Slic3r-masterPE/build/googletest-src/include/gtest/gtest.h: In instantiation of ‘testing::AssertionResult testing::internal::CmpHelperEQ(const char*, const char*, const T1&, const T2&) [with T1 = long unsigned int; T2 = int]’:
/home/mperino/3d/Software/Slic3r-masterPE/build/googletest-src/include/gtest/gtest.h:1484:23:   required from ‘static testing::AssertionResult testing::internal::EqHelper<lhs_is_null_literal>::Compare(const char*, const char*, const T1&, const T2&) [with T1 = long unsigned int; T2 = int; bool lhs_is_null_literal = false]’
/home/mperino/3d/Software/Slic3r-masterPE/src/libnest2d/tests/test.cpp:829:5:   required from here
/home/mperino/3d/Software/Slic3r-masterPE/build/googletest-src/include/gtest/gtest.h:1448:16: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
   if (expected == actual) {
       ~~~~~~~~~^~~~~~~~~
src/libslic3r/CMakeFiles/libslic3r.dir/build.make:66: recipe for target 'src/libslic3r/libslic3r_pch/pchheader.hpp.gch/.c++' failed
make[2]: *** [src/libslic3r/libslic3r_pch/pchheader.hpp.gch/.c++] Error 1
CMakeFiles/Makefile2:1311: recipe for target 'src/libslic3r/CMakeFiles/libslic3r.dir/all' failed
make[1]: *** [src/libslic3r/CMakeFiles/libslic3r.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 55%] Linking CXX executable tests_clipper_nlopt
[ 55%] Built target tests_clipper_nlopt
Makefile:162: recipe for target 'all' failed
make: *** [all] Error 2

supermerill, 2019-02-26:
corrected

now you can
git fetch --all
git reset --hard origin/masterPE

mperino, 2019-02-27:
$ git reset --hard origin/masterPE
HEAD is now at e0812a714 bugfix #47

[ 59%] Building CXX object src/libslic3r/CMakeFiles/libslic3r.dir/Fill/Fill.cpp.o
[ 59%] Building CXX object src/libslic3r/CMakeFiles/libslic3r.dir/Fill/Fill3DHoneycomb.cpp.o
/home/mperino/3D/Software/supermerril/Slic3r/src/libslic3r/Fill/Fill.cpp: In function ‘void Slic3r::make_fill(Slic3r::LayerRegion&, Slic3r::ExtrusionEntityCollection&)’:
/home/mperino/3D/Software/supermerril/Slic3r/src/libslic3r/Fill/Fill.cpp:316:15: warning: unused variable ‘flow_percent’ [-Wunused-variable]
         float flow_percent = 1;
               ^~~~~~~~~~~~
/home/mperino/3D/Software/supermerril/Slic3r/src/libslic3r/Fill/Fill.cpp:38:13: warning: variable ‘infill_flow’ set but not used [-Wunused-but-set-variable]
     Flow    infill_flow             = layerm.flow(frInfill);
             ^~~~~~~~~~~
[ 59%] Building CXX object src/libslic3r/CMakeFiles/libslic3r.dir/Fill/FillBase.cpp.o
In file included from /home/mperino/3D/Software/supermerril/Slic3r/src/libslic3r/Fill/FillBase.cpp:17:0:
/home/mperino/3D/Software/supermerril/Slic3r/src/libslic3r/Fill/FillSmooth.hpp:52:10: error: extra qualification ‘Slic3r::FillSmooth::’ on member ‘performSingleFill’ [-fpermissive]
     void FillSmooth::performSingleFill(const int idx, ExtrusionEntityCollection &eecroot, const Surface &srf_source,
          ^~~~~~~~~~
/home/mperino/3D/Software/supermerril/Slic3r/src/libslic3r/Fill/FillSmooth.hpp:54:10: error: extra qualification ‘Slic3r::FillSmooth::’ on member ‘fillExPolygon’ [-fpermissive]
     void FillSmooth::fillExPolygon(const int idx, ExtrusionEntityCollection &eec, const Surface &srf_to_fill,
          ^~~~~~~~~~
[ 60%] Building CXX object src/libslic3r/CMakeFiles/libslic3r.dir/Fill/FillConcentric.cpp.o
/home/mperino/3D/Software/supermerril/Slic3r/src/libslic3r/Fill/FillConcentric.cpp: In member function ‘virtual void Slic3r::FillConcentricWGapFill::fill_surface_extrusion(const Slic3r::Surface*, const Slic3r::FillParams&, Slic3r::ExtrusionEntitiesPtr&)’:
/home/mperino/3D/Software/supermerril/Slic3r/src/libslic3r/Fill/FillConcentric.cpp:96:18: warning: unused variable ‘init_spacing’ [-Wunused-variable]
         coordf_t init_spacing = this->spacing;
                  ^~~~~~~~~~~~
src/libslic3r/CMakeFiles/libslic3r.dir/build.make:437: recipe for target 'src/libslic3r/CMakeFiles/libslic3r.dir/Fill/FillBase.cpp.o' failed
make[2]: *** [src/libslic3r/CMakeFiles/libslic3r.dir/Fill/FillBase.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:1311: recipe for target 'src/libslic3r/CMakeFiles/libslic3r.dir/all' failed
make[1]: *** [src/libslic3r/CMakeFiles/libslic3r.dir/all] Error 2
Makefile:162: recipe for target 'all' failed
make: *** [all] Error 2

supermerill, 2019-02-27:
corrected.

note: you should be able to compile with the -fpermissive option.

mperino, 2019-02-27:
Closer.. 

I edited the file "CMakeCache.txt" to read:

//Flags used by the compiler during all build types.
CMAKE_CXX_FLAGS:STRING="-fpermissive"

gets further.. and now an error "error: cannot bind non-const lvalue reference of type ‘Slic3r::Line&’ to an rvalue of type ‘Slic3r::Line" in PerimeterGenerator.cpp

 
[ 71%] Building CXX object src/libslic3r/CMakeFiles/libslic3r.dir/PerimeterGenerator.cpp.o
/home/mperino/3d/Software/Slic3r-masterPE/Slic3r/src/libslic3r/PerimeterGenerator.cpp: In member function ‘void Slic3r::PerimeterGenerator::process()’:
/home/mperino/3d/Software/Slic3r-masterPE/Slic3r/src/libslic3r/PerimeterGenerator.cpp:151:37: warning: unused variable ‘numploy’ [-Wunused-variable]
                                 int numploy = 0;
                                     ^~~~~~~
/home/mperino/3d/Software/Slic3r-masterPE/Slic3r/src/libslic3r/PerimeterGenerator.cpp:203:51: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
                                 for (int i = 0; i < last.size();i++) {
                                                 ~~^~~~~~~~~~~~~
/home/mperino/3d/Software/Slic3r-masterPE/Slic3r/src/libslic3r/PerimeterGenerator.cpp:423:43: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
                         for (int j = 0; j < contours[t].size(); ++j) {
                                         ~~^~~~~~~~~~~~~~~~~~~~
/home/mperino/3d/Software/Slic3r-masterPE/Slic3r/src/libslic3r/PerimeterGenerator.cpp: In member function ‘Slic3r::ExtrusionLoop Slic3r::PerimeterGenerator::_extrude_and_cut_loop(const Slic3r::PerimeterGeneratorLoop&, Slic3r::Point, const Slic3r::Line&) const’:
/home/mperino/3d/Software/Slic3r-masterPE/Slic3r/src/libslic3r/PerimeterGenerator.cpp:874:45: error: cannot bind non-const lvalue reference of type ‘Slic3r::Line&’ to an rvalue of type ‘Slic3r::Line’
                 coord_t dot = direction.dot(Line(direction_polyline.points.back(), direction_polyline.points.front()));
                                             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/mperino/3d/Software/Slic3r-masterPE/Slic3r/src/libslic3r/Polygon.hpp:7:0,
                 from /home/mperino/3d/Software/Slic3r-masterPE/Slic3r/src/libslic3r/BoundingBox.hpp:6,
                 from /home/mperino/3d/Software/Slic3r-masterPE/Slic3r/build/src/libslic3r/libslic3r_pch/pchheader.hpp:105:
/home/mperino/3d/Software/Slic3r-masterPE/Slic3r/src/libslic3r/Line.hpp:50:13: note:   initializing argument 1 of ‘coord_t Slic3r::Line::dot(Slic3r::Line&) const’
     coord_t Line::dot(Line &l2) const { return vector().dot(l2.vector()); }
             ^~~~
/home/mperino/3d/Software/Slic3r-masterPE/Slic3r/src/libslic3r/PerimeterGenerator.cpp:912:45: error: cannot bind non-const lvalue reference of type ‘Slic3r::Line&’ to an rvalue of type ‘Slic3r::Line’
                 coord_t dot = direction.dot(Line(direction_polyline.points.back(), direction_polyline.points.front()));
                                             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/mperino/3d/Software/Slic3r-masterPE/Slic3r/src/libslic3r/Polygon.hpp:7:0,
                 from /home/mperino/3d/Software/Slic3r-masterPE/Slic3r/src/libslic3r/BoundingBox.hpp:6,
                 from /home/mperino/3d/Software/Slic3r-masterPE/Slic3r/build/src/libslic3r/libslic3r_pch/pchheader.hpp:105:
/home/mperino/3d/Software/Slic3r-masterPE/Slic3r/src/libslic3r/Line.hpp:50:13: note:   initializing argument 1 of ‘coord_t Slic3r::Line::dot(Slic3r::Line&) const’
     coord_t Line::dot(Line &l2) const { return vector().dot(l2.vector()); }
             ^~~~
/home/mperino/3d/Software/Slic3r-masterPE/Slic3r/src/libslic3r/PerimeterGenerator.cpp: In member function ‘bool Slic3r::PerimeterGeneratorLoop::is_internal_contour() const’:
/home/mperino/3d/Software/Slic3r-masterPE/Slic3r/src/libslic3r/PerimeterGenerator.cpp:1228:5: warning: this ‘for’ clause does not guard... [-Wmisleading-indentation]
     for (const PerimeterGeneratorLoop &loop : this->children)
     ^~~
/home/mperino/3d/Software/Slic3r-masterPE/Slic3r/src/libslic3r/PerimeterGenerator.cpp:1231:9: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the ‘for’
         return true;
         ^~~~~~
[ 71%] Building CXX object src/libslic3r/CMakeFiles/libslic3r.dir/PlaceholderParser.cpp.o
src/libslic3r/CMakeFiles/libslic3r.dir/build.make:1399: recipe for target 'src/libslic3r/CMakeFiles/libslic3r.dir/PerimeterGenerator.cpp.o' failed
make[2]: *** [src/libslic3r/CMakeFiles/libslic3r.dir/PerimeterGenerator.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
/home/mperino/3d/Software/Slic3r-masterPE/Slic3r/src/libslic3r/PlaceholderParser.cpp: In function ‘std::__cxx11::string Slic3r::process_macro(const string&, Slic3r::client::MyContext&)’:
/home/mperino/3d/Software/Slic3r-masterPE/Slic3r/src/libslic3r/PlaceholderParser.cpp:1248:10: warning: unused variable ‘res’ [-Wunused-variable]
     bool res = phrase_parse(iter, end, macro_processor_instance(&context), space, output);
          ^~~
/home/mperino/3d/Software/Slic3r-masterPE/Slic3r/src/libslic3r/PlaceholderParser.cpp: In instantiation of ‘Slic3r::client::macro_processor<Iterator>::macro_processor() [with Iterator = __gnu_cxx::__normal_iterator<const char*, std::__cxx11::basic_string<char> >]’:
[... cut ...]
/home/mperino/3d/Software/Slic3r-masterPE/Slic3r/src/libslic3r/PlaceholderParser.cpp:1242:33:   required from here
/home/mperino/3d/Software/Slic3r-masterPE/Slic3r/src/libslic3r/PlaceholderParser.cpp:524:32: warning: unused variable ‘mask’ [-Wunused-variable]
             const std::string *mask     = nullptr;
                                ^~~~
CMakeFiles/Makefile2:1311: recipe for target 'src/libslic3r/CMakeFiles/libslic3r.dir/all' failed
make[1]: *** [src/libslic3r/CMakeFiles/libslic3r.dir/all] Error 2
Makefile:162: recipe for target 'all' failed
make: *** [all] Error 2

supermerill, 2019-02-27:
oups, i accidentally reverted the previous changes. Should be better now.

mperino, 2019-02-27:
Now onto errors in PrintObject.cpp :)

Fetching origin
remote: Enumerating objects: 5, done.
remote: Counting objects: 100% (5/5), done.
remote: Total 5 (delta 4), reused 5 (delta 4), pack-reused 0
Unpacking objects: 100% (5/5), done.
From https://github.com/supermerill/Slic3r
 + 46e6d3e97...626f8af98 masterPE   -> origin/masterPE  (forced update)

deleted build and re-ran:
cmake -DSLIC3R_WX_STABLE=1 ..
make -j3

[ 74%] Building CXX object src/libslic3r/CMakeFiles/libslic3r.dir/PrintObject.cpp.o
[ 74%] Building CXX object src/libslic3r/CMakeFiles/libslic3r.dir/PrintRegion.cpp.o
/home/mperino/3d/Software/Slic3r-masterPE/Slic3r/src/libslic3r/PrintObject.cpp: In member function ‘void Slic3r::PrintObject::detect_surfaces_type()’:
/home/mperino/3d/Software/Slic3r-masterPE/Slic3r/src/libslic3r/PrintObject.cpp:865:41: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
     for (int idx_region = 0; idx_region < this->region_volumes.size(); ++ idx_region) {
                              ~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/mperino/3d/Software/Slic3r-masterPE/Slic3r/src/libslic3r/PrintObject.cpp: In lambda function:
/home/mperino/3d/Software/Slic3r-masterPE/Slic3r/src/libslic3r/PrintObject.cpp:899:34: warning: unused variable ‘under_lower_layer’ [-Wunused-variable]
                     Layer       *under_lower_layer = (idx_layer > 1) ? this->layers()[idx_layer - 2] : nullptr;
                                  ^~~~~~~~~~~~~~~~~
/home/mperino/3d/Software/Slic3r-masterPE/Slic3r/src/libslic3r/PrintObject.cpp: In member function ‘void Slic3r::PrintObject::process_external_surfaces()’:
/home/mperino/3d/Software/Slic3r-masterPE/Slic3r/src/libslic3r/PrintObject.cpp:1053:28: warning: unused variable ‘region’ [-Wunused-variable]
         const PrintRegion &region = *m_print->regions()[region_id];
                            ^~~~~~
/home/mperino/3d/Software/Slic3r-masterPE/Slic3r/src/libslic3r/PrintObject.cpp: In lambda function:
/home/mperino/3d/Software/Slic3r-masterPE/Slic3r/src/libslic3r/PrintObject.cpp:1275:40: warning: unused variable ‘neighbor_layer’ [-Wunused-variable]
                                 Layer &neighbor_layer = *m_layers[n];
                                        ^~~~~~~~~~~~~~
/home/mperino/3d/Software/Slic3r-masterPE/Slic3r/src/libslic3r/PrintObject.cpp: In member function ‘void Slic3r::PrintObject::replaceSurfaceType(Slic3r::SurfaceType, Slic3r::SurfaceType, Slic3r::SurfaceType)’:
/home/mperino/3d/Software/Slic3r-masterPE/Slic3r/src/libslic3r/PrintObject.cpp:1597:28: warning: unused variable ‘region’ [-Wunused-variable]
         const PrintRegion &region = *m_print->regions()[region_id];
                            ^~~~~~
/home/mperino/3d/Software/Slic3r-masterPE/Slic3r/src/libslic3r/PrintObject.cpp: In member function ‘void Slic3r::PrintObject::_offsetHoles(float, Slic3r::LayerRegion*)’:
/home/mperino/3d/Software/Slic3r-masterPE/Slic3r/src/libslic3r/PrintObject.cpp:1909:42: error: cannot bind non-const lvalue reference of type ‘Slic3r::ExPolygons& {aka std::vector<Slic3r::ExPolygon>&}’ to an rvalue of type ‘Slic3r::ExPolygons {aka std::vector<Slic3r::ExPolygon>}’
         ExPolygons& polys = to_expolygons(std::move(layerm->slices.surfaces));
                             ~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/mperino/3d/Software/Slic3r-masterPE/Slic3r/src/libslic3r/PrintObject.cpp: In member function ‘void Slic3r::PrintObject::discover_horizontal_shells()’:
/home/mperino/3d/Software/Slic3r-masterPE/Slic3r/src/libslic3r/PrintObject.cpp:2321:46: warning: suggest parentheses around comparison in operand of ‘|’ [-Wparentheses]
                     if (surface.surface_type == stPosInternal | stDensSparse)
                         ~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~
/home/mperino/3d/Software/Slic3r-masterPE/Slic3r/src/libslic3r/PrintObject.cpp:2358:56: warning: suggest parentheses around comparison in operand of ‘&’ [-Wparentheses]
                 size_t solid_layers = (type & stPosTop == stPosTop) ? region_config.top_solid_layers.value : region_config.bottom_solid_layers.value;
                                               ~~~~~~~~~^~~~~~~~~~~
/home/mperino/3d/Software/Slic3r-masterPE/Slic3r/src/libslic3r/PrintObject.cpp:2359:47: warning: suggest parentheses around comparison in operand of ‘&’ [-Wparentheses]
                 for (int n = (type & stPosTop == stPosTop) ? i - 1 : i + 1; std::abs(n - i) < solid_layers; (type & stPosTop == stPosTop) ? --n : ++n) {
                                      ~~~~~~~~~^~~~~~~~~~~
/home/mperino/3d/Software/Slic3r-masterPE/Slic3r/src/libslic3r/PrintObject.cpp:2359:126: warning: suggest parentheses around comparison in operand of ‘&’ [-Wparentheses]
                 for (int n = (type & stPosTop == stPosTop) ? i - 1 : i + 1; std::abs(n - i) < solid_layers; (type & stPosTop == stPosTop) ? --n : ++n) {
                                                                                                                     ~~~~~~~~~^~~~~~~~~~~
/home/mperino/3d/Software/Slic3r-masterPE/Slic3r/src/libslic3r/PrintObject.cpp: In member function ‘void Slic3r::PrintObject::combine_infill()’:
/home/mperino/3d/Software/Slic3r-masterPE/Slic3r/src/libslic3r/PrintObject.cpp:2526:95: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
                 if (current_height + layer->height >= nozzle_diameter + EPSILON || num_layers >= every) {
[... cut ...] ~~~
src/libslic3r/CMakeFiles/libslic3r.dir/build.make:1633: recipe for target 'src/libslic3r/CMakeFiles/libslic3r.dir/PrintObject.cpp.o' failed
make[2]: *** [src/libslic3r/CMakeFiles/libslic3r.dir/PrintObject.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:1311: recipe for target 'src/libslic3r/CMakeFiles/libslic3r.dir/all' failed
make[1]: *** [src/libslic3r/CMakeFiles/libslic3r.dir/all] Error 2
Makefile:162: recipe for target 'all' failed
make: *** [all] Error 2

supermerill, 2019-02-27:
hope the last one will be the last.
maybe i should set up a linux build, it will be more efficient.

mperino, 2019-03-01:
Sorry for the delay.  Yes, it worked!  Thank you for all the fixes along the way!

mperino, 2019-03-01:
Perhaps it's a silly Question, but where is ironing in the options (or is it always on?).

mperino, 2019-03-04:
Thank you for ALL your hard work.  Ironing turned out to be a huge help for a very detailed model that our FRC kids are printing (0.3 nozzle, 0.05 layer height).  We dont have access to an SLA, and they needed to print as detailed as possible for a "Dubloon" that they are cold casting.
Thanks!  Cant wait for Ironing to get into the main PE builds!

[image]

closed 2019-03-14 by supermerill