supermerill/SuperSlicer #49: Settings UI is slow

Murloc992, 2019-02-28:
**Version**
1.42.0 - alpha6.0

**Operating system type + version**
Windows 10 64bit

**3D printer brand / version + firmware version (if known)**
Unrelated

**_Behavior_**

**Describe the problem**
UI is slow when going inbetween UI elements with mouse. For example if I click on extrusion widths, they hang entire slic3r for a second or two. Especially apparent with several instances launched. Setting up a 1mm nozzle took me close to 10 minutes because of constant hanging and dialogs that ask me to make sure my nozzle is really 1mm. 😄 

**Steps needed to reproduce the problem**
Turn on Slic3r and click around extrusion width inputs or other settings.

**Expected Results**
UI is as responsive as it was in Slic3rPE 1.42

**Actual Results**
Hanging on every input field click..

Maybe it's a debug build of the application?

supermerill, 2019-02-28:
it's also bugging me. I will investigate why there is a difference between me and prusa.

supermerill, 2019-03-14:
I checked my 1.42.0-alpha7.0 and prusa 1.42.0-alpha7 release and i can't see any difference. If you spot an obvious one, please re-open this (or create a new one) and write down how to reproduce (with the needed files).

closed 2019-03-14 by supermerill