supermerill/SuperSlicer #50: Print time calculation issue with cooling options

ewb-git, 2019-03-03:
I figure I would mention this to help improve the slic3r.

I talked with you on discord, I had my infill printing very slowly, you wrote “It's possible if your filament cooling time is greater than the layer time. It will slow down the infill and try to let the perimeter at the same speed if possible.”

I was in fact playing with cooling layer times. I raised/lose the various values to try to increase the print time, with the goal of slowing down a particular print/improving quality time.

I’d increase the value, and slice, and look at the print time. However, what needed up being a ~2 hour print time difference due to slow infill speed/wait time, the slic3r only showed an increased print time of roughly 3 minutes.

So...basically the print time calculation isn’t accurate taking into account these changes. A 2 minute estimated print time increase could actually result in 2 hours longer print time on a small object. On a larger object, the time difference could be days.

supermerill, 2019-03-03:
can't resolve these sort of issue without the project file. you can zip it and darg-drop it, or upload it to a filesharing service before posting the link here.

ewb-git, 2019-03-03:
Alright, I’m not sure if I saved the config for the particular print but I think I know what was causing it more or less. Writing from my phone at the moment...

supermerill, 2020-07-03:
things have changed now, re-open it if the issue happens again.

closed 2020-07-03 by supermerill