supermerill/SuperSlicer #51: Slicing sequential objects skirt is generated around all copies of object

tomasvilda, 2019-03-05:
Version
1.4.20 alpha 6

Operating system type + version
Windows 10

Behavior
Slicing sequential objects skirt is generated around all objects and not around every sequential object.

Steps needed to reproduce the problem
Create a copy of the object and choose complete individual objects in sequential printing.

Expected Results
Skirt should be around every copy of the object.

Actual Results
One skirt around all sequential objects.

supermerill, 2019-03-05:
but you won't complain if a skirt is passing by an already finished part?

Maybe automatically deactivating multi-layers skirt is the way to go.

tomasvilda, 2019-03-05:
There is notification when multiple objects are too close to each other, the same way user could be notified about overlapping skirts. On Cura every copy has it's own skirt.

Without skirt I get mess on the second copy of the object when printing TPU. Skirt cleans the way before printing the second object.

supermerill, 2019-03-05:
I'll see if i can do something quick. Currently skirt &brim are extruded before everything, so it may be big task.

tomasvilda, 2019-03-14:
Using alpha 7.0, but the result is the same, one skirt around all individual objects.

supermerill, 2019-03-15:
[image]

that's not what you want?

tomasvilda, 2019-03-15:
Yes that's what I want and you showed workaround, just add as many objects as I want on plate. But that should work on instances also. Because if I add object and add instances I get one skirt around all instances.

closed 2019-07-08 by supermerill