supermerill/SuperSlicer #52: "Vector <T> too long" when slicing certain objects.

repvik, 2019-03-08:
Version Slic3r++_1.42.0-alpha6.0_190221

Operating system type + version
Windows 10, all updates. 

3D printer brand / version + firmware version (if known)
MK3 Bear upgrade

Behavior
Slic3r++ crashes while slicing.

Describe the problem
Slicing certain objects throws an "Vector <T> too long" during slicing. Slic3r++ has to be closed and reopened to work again. Other objects work fine. Objects also slice fine in Slic3rPE.

Steps needed to reproduce the problem

I've attached three files from the same thingiverse thing: https://www.thingiverse.com/thing:3469271
The extruder parts fail to slice, throwing the error message. The carriage slices fine. 

Expected Results
A complete slice. 

Actual Results
Aborted slice, error message and need to close slic3r.

Screenshots from Slic3r preview are preferred

# Project File (.3MF) (or stl+config) where problem occurs
### file-> save as
### zip it
### drag it here
[files.zip](https://github.com/supermerill/Slic3r/files/2945560/files.zip)
[config.zip](https://github.com/supermerill/Slic3r/files/2945566/config.zip)

repvik, 2019-03-08:
If I disable generating supports, I can slice it fine. So it occurs during support generation.

closed 2019-03-14 by supermerill

digitall-it, 2023-11-24:
Same problem, Orca Slicer v1.8.0-beta2, disabling support generation removes the problem. The message on screen just says "vector" but it is presumable it is the same. MacOS Sonoma, stock profiles.

commit 8c8f822b (2019-03-08): fix #52 (hopefully)