supermerill/SuperSlicer #56: Not starting pre-compiled version, MSVCP120.dll is missing

birtalanelod, 2019-03-09:
Version : Slic3r++_1.42.0-alpha6.0_190221
Not starting pre-compiled version , 
MSVCP120.dll is reported missing
[image]

supermerill, 2019-03-10:
oups.
You can steal it from slic3rPE or microsoft for now.

closed 2019-03-14 by supermerill

birtalanelod, 2019-03-14:
the solution was to install Visual C++ Redistributable Packages for Visual Studio 2013 from this link:
https://www.microsoft.com/en-us/download/details.aspx?id=40784

sorry for posting as an issue, I could just Google it, but instead I went crazy (because  slic3r PE was working)

supermerill, 2019-03-15:
no problem, i'm glad you post it, i may have forgotten to add it without you.