supermerill/SuperSlicer #57: [Feature Request] Coasting

dot-bob, 2019-03-19:
I have a couple of printers that have a E3D volcano with a nozzle size of 1mm and greater.  The retraction options provided by slic3r doesn't have the ability to adjust the coasting under retraction options which is very helpful for larger nozzles.

It would be nice to have this option to provide more tuning flexibility for larger nozzle sizes.

supermerill, 2019-03-19:
what is coasting?

dot-bob, 2019-03-20:
The basic idea of costing is to replace the last part of an extrusion path with a travel move.  Other words quit extruding a set distance before a retraction operation.  On larger nozzle that have a tendency to ooze for a while after releasing pressure you use a combination of coasting and extra restart distance to control oozing.

See the following for reference:
[Cura: Enable Coasting](https://ultimaker.com/en/resources/20426-experimental)
[Simplify3D: Coasting Distance](https://forum.simplify3d.com/viewtopic.php?t=1980)
[Ideamaker: Coasting Distance](https://www.raise3d.com/blogs/news/ideamaker-3-1-0)

supermerill, 2019-03-20:
I looked at the code and it seems it's a very time-consuming feature.
If someone is motivated to code it, no problem for merging it.
For me, as i didn't need it right now, i will wait until i have nothing else to do.

MisterPrintf, 2019-04-15:
Have you tried the wipe on retract option? I have a drippy PLA right now, so I do 70mm/sec retraction with 50% of the retract accomplished before the wipe. Note that you'll need to set an extra prime after retraction.

dot-bob, 2019-04-26:
With a 1.2mm nozzle wipe with retract can actually make it worse by making the seam more pronounced.  Depending on the material and moisture content it can continue to flow for a while after retraction.  PETG is a good example of this.

guestisp, 2019-04-26:
+1 for this but I agree, it doesn't seam trivial feature to add

albert2004, 2019-08-30:
What is expected difference (except GCode vs Firmware) with Marlin LINEAR_ADVANCE feature?
http://marlinfw.org/docs/features/lin_advance.html

wavexx, 2019-08-30:
Linear advance should completely remove the need for coasting by compensating the excess pressure directly in hardware and, if your printer has it, you should definitely favor linear advance to coasting. A properly tuned LA should also remove the need for wipe-during-retract as well (which has a tendency to overheat small features).

On larger nozzles and/or less viscous materials though you might still get oozing. Coasting won't really help there. In these cases what I would really like to see is the "combing" feature of cura, which is basically an enhanced "wipe+retract" where the final wipe is performed towards the interior of the volume.

supermerill, 2019-08-30:
Linear advance is useful to modelize the pressure model: the pressure doesn't build up at the moment you unretract, because of some springyness. So It takes some time, and that's what LINEAR_ADVANCE try to compensate.

If the printer ahs a perfect model of the pressure, there are something else that can make the plastic flow outside of the nozzle: the gravity. Because the viscosity isn't big enough for the size of the nozzle, it will slowly flow outside of it. 
 - One way to compensate that is to extrude with a lower temperature, if it's possible (it can creates adhesion problems). 
 - One other way is to let the plastic ooze to decrease the mass of molten palstic. When the mass is low enough, the surface tension can retain the rest inside the nozzle/heat bloc. A bigger unretraction will be needed to fill the nozzle/heat bloc again. I think that's what coasting is doing. Problem is, it's difficult to know the amount of plastic that ooze (it may need to begin with isparse infill to compensate for the possibility of under/overextrusion at startup), it's difficult to know how many time to wait is needed, it's difficult to find an area where the nozzle can ooze (if the area is solid, it's impossible), and a wipe is needed after that to separate the string from the nozzle (and that may create other problems).

Also, if there are a little seal at the tip of the hotend, the retraction may be able to "suck" the molten plastic from the nozzle into the heatBlock, if the viscosity is high enough. (I disn't see any printer with that, it's just an idea).

dot-bob, 2019-08-30:
> Also, if there are a little seal at the tip of the hotend, the retraction may be able to "suck" the molten plastic from the nozzle into the heatBlock, if the viscosity is high enough. (I didn't see any printer with that, it's just an idea).
> 

I have worked on a couple of different implementations of this idea.  I have made a couple of different designs of hotends and heatbreaks to seal around the filament.  The idea is to act similarly to a syringe so when the filament is retracted it draws the molten filament back in the nozzle.  There is a slight but noticeable improvement but it didn't eliminate ooze entirely.  By far the simplest implementation I tried was to take a 4mm OD x 1mm ID [o-ring](https://www.amazon.com/gp/product/B01MTNJ7FI) insert it into the top of an E3D all metal 1.75mm heatbreak and use a ptfe tube and the bowden coupler to hold and sandwich it in the top recess of the heatbreak.  You just need to make sure the filament is clean.

supermerill, 2020-07-03:
i've added an "extra wipe" tat is a bit like a coasting.

VanessaE, 2020-09-01:
I'd like to add a remark here that the extra wipe is not like coast at all, in that it only stops the extrusion after the last printed segment, making that feature useful only on parts that are large enough to have sparse infill in which to deposit the excess plastic (and to then re-prime the nozzle).

Coast on the other hand stops extruding *during* the last printed segment, letting the residual pressure draw the last of it, so that you only need a small amount of solid infill, gap fill,  or an inner perimeter to re-prime the nozzle after the coast ends.

supermerill, 2020-09-01:
oh, ok.

For doing that, I need to implement a pressure model and use it on every speed change, retraction, and unretraction.

VanessaE, 2020-09-01:
Well, as far as I know, it doesn't get into the whole dynamic pressure thing; coasting is just a fixed (configurable) length and only applies to the last print move before a travel, if that line would be visible.  You just shorten the line by N millemeters, then insert an N mm travel move (at the *print* speed) to make up for that.

madsi1m, 2020-09-20:
Hi @supermerill, first off great work, I discovered this app 2 days ago and loving it! My fav features is the turn the fan on early and single perimeter on the top layer (which I used to manually do in prusaslicer)

Coasting is something I've been waiting for a long time.

In Marlin I have tuned LINEAR_ADVANCE and it has improved things a lot but there seems to still be a need for coasting.
In Klipper, I have tuned pressure_advance and found it better than LINEAR_ADVANCE but there is still that tiny bit of improvement I get with coasting.

I like to just add, at least in my case, I only really need this for the last line in my external perimeters. (can I scope creap a tickbox?)

As a basic test try printing a test cube and one corner (where the seam is) is always slightly lumpier than the others.

Keep up the good work!

robschwieb, 2020-10-01:
Just chiming in to say that I'd love to have this feature. It's the only thing keeping me from using SuperSlicer exclusively. Folks over at PrusaSlicer have been asking for this feature for years. [Issue 246](https://github.com/prusa3d/PrusaSlicer/issues/246)

With that said, great work on SuperSlicer and many thanks! It's fixed so many things that annoy me about PrusaSlicer all while keeping the very efficient Slic3r UI.

CCS86, 2020-12-09:
Another vote for coasting. It is definitely a useful feature.

Fabface, 2021-12-10:
Hi, two things are not working as I would expect with the seam gap to work:
1. if you set the gap in mm instead of % the setting is just ignored and there is no gap.
2. It actually leaves a real gap, instead of travelling to the start point of the perimeter to close the loop without extruding  it travels right from the start of the gap to the next perimeter (its possible to force the correct behaviour by using "extra wipe for external perimeters" however internal perimeters are left with a gap)

I added the expected path in green below, blue is the actual path. 

[image]

supermerill, 2021-12-11:
I wonder if it's really useful to apply it also for inner perimeters (the yellow ones)

CCS86, 2021-12-11:
I think it is.

When all seams are aligned like that, any seam bulge will be a cumulative effect; where the prior seam bulge encroaches on the next path. 

Applying to all perimeters would likely allow a much smaller value.

closed 2021-12-22 by supermerill

fishcu, 2021-12-31:
Supermerill, thank you so much for your excellent work on this slicer!

Regarding the coasting feature, there's a behavior that I would expect differently.
Namely, when printing with infill, the coasting is only done on perimeters, but not on infill.
However, usually the infill is the last extrusion before a retraction.
This means that leaky nozzles will print the perimeters with coasting, then print the infill without coasting, then retract and travel to another part of the object. However, the pressure before the travel is not released due to the coasting because it builds up during the infill printing again.

What I would expect instead is that coasting is applied before **every** retraction!

What I am doing now is to enable "infill before perimeters" to avoid this issue. However, this workaround brings other problems such as the first layer infill being stuck on the plate without perimeters which usually give it more stability.

Do you agree with my (theoretical) explanation above, or is there another workaround?

Here is a screenshot with the default perimeters before infill ordering:
[image]

Here is a screenshot with "infill before perimeters":
[image]

supermerill, 2022-01-04:
can't improve on it right now.