supermerill/SuperSlicer #60: Crash when slicing excavator model

MisterPrintf, 2019-03-25:
Version
Early fan edition

Operating system type + version
Windows 7 home

3D printer brand / version + firmware version (if known)
Creality Ender-3 (running Creality's modified version of Marlin)

Behavior
Slic3r crashes when slicing the 'excavator_extended' model found here:
https://www.thingiverse.com/thing:2719434

Steps needed to reproduce the problem
Slice the model

Expected Results
Slic3r should not crash.

Actual Results
Slic3r crashes.

Note that I have changed my slicing settings a couple times with no success so you shouldn't need the profile.

supermerill, 2019-03-25:
could not reproduce.

please zip & post your project (.3mf) file.

[image]

MisterPrintf, 2019-03-28:
My apologies for the delay. As it turns out, the crash only occurs when supports are enabled (not sure why you'd want this, but it does crash).

I've attached a zip with the STL and the config file.

MisterPrintf, 2019-03-28:
My apologies for the delay. I've attached a config file as requested. Also included is the .stl though it looks like you've got the right one already.

On a side note, it appears the crash only occurs when support generation is turned on.

supermerill, 2019-03-28:
I can't find your files, note that you may have to zip them first.

MisterPrintf, 2019-03-29:
Sorry about that. I forgot that IE likes being picky with Github.

Here are the files. (I hope).
[config.zip](https://github.com/supermerill/Slic3r/files/3020859/config.zip)

supermerill, 2019-03-29:
I forgot to convert a code path from unsafe cast to generic oop. May be caused by my new bottom pattern options for supports.

MisterPrintf, 2019-06-17:
Just tested with the latest version and it works! Closing the issue now.

closed 2019-06-17 by MisterPrintf