supermerill/SuperSlicer #64: assert "item" failed in SelectItem(): SelectItem(): invalid tree item

guestisp, 2019-05-09:
### Version
https://github.com/supermerill/Slic3r/commit/6df17e1dd160ff617783058a19c0f1b1e47ef20d

### Operating system type + version
Distributor ID:	Ubuntu
Description:	Ubuntu 19.04
Release:	19.04
Codename:	disco

ASSERT INFO:
../src/generic/treectlg.cpp(2216): assert "item" failed in SelectItem(): SelectItem(): invalid tree item

BACKTRACE:
[1] wxGenericTreeCtrl::SelectItem(wxTreeItemId const&, bool)
[2] Slic3r::GUI::Tab::update_page_tree_visibility()
[3] Slic3r::GUI::Tab::update_visibility()
[4] Slic3r::GUI::Tab::load_current_preset()
[5] Slic3r::GUI::GUI_App::load_current_presets()
[6] Slic3r::GUI::GUI_App::OnInit()
[7] wxEntry(int&, wchar_t**)
[8] Slic3r::CLI::run(int, char**)
[9] main
[10] __libc_start_main
[11] _start

supermerill, 2019-05-11:
config conflict, i suppose. Try with a clean setting folder or build without asserts.

guestisp, 2019-05-11:
Conflict with Slic3r Prusa config? Can't you change the config dir of your fork so that both instances can be used?

supermerill, 2019-05-16:
I've done that, but the config can "confict" if it's from an earlier version.
I disabled some assert that were too anoying, but i leave these ones, as they seems legit.
I spent hours to see what's wrong, and it's mainly "hey, you don't have this new option in your config, it's not normal".

guestisp, 2019-05-16:
In my case, it was a simple install, i've not copied an older config to your fork. It was fresh

supermerill, 2019-05-16:
i'll look at it.

supermerill, 2019-12-12:
Should be okay now, i didn't had these in a long time.

closed 2019-12-12 by supermerill