supermerill/SuperSlicer #65: assert "item.IsOk()" failed in GetItemText(): invalid tree item

guestisp, 2019-05-09:
### Version
https://github.com/supermerill/Slic3r/commit/6df17e1dd160ff617783058a19c0f1b1e47ef20d

### Operating system type + version
Distributor ID:	Ubuntu
Description:	Ubuntu 19.04
Release:	19.04
Codename:	disco

### Behavior
Just open the program and you'll get the following backtrace:
```
ASSERT INFO:
../src/generic/treectlg.cpp(1129): assert "item.IsOk()" failed in GetItemText(): invalid tree item

BACKTRACE:
[1] wxGenericTreeCtrl::GetItemText(wxTreeItemId const&) const
[2] Slic3r::GUI::Tab::update_changed_tree_ui()
[3] Slic3r::GUI::GUI_App::update_mode()
[4] Slic3r::GUI::GUI_App::OnInit()
[5] wxEntry(int&, wchar_t**)
[6] Slic3r::CLI::run(int, char**)
[7] main
[8] __libc_start_main
[9] _start
```

supermerill, 2019-05-11:
Maybe a configuration mismatch. Most of the time, it's nothing bad (just an option that isn't here or removed). Move all your current config to get a blank state or build without asserts.

supermerill, 2019-12-12:
Should have been resolved by prusa now.

closed 2019-12-12 by supermerill