supermerill/SuperSlicer #66: Coordinate outside allowed range

guestisp, 2019-05-09:
### Version
https://github.com/supermerill/Slic3r/commit/6df17e1dd160ff617783058a19c0f1b1e47ef20d

### Operating system type + version
Distributor ID:	Ubuntu
Description:	Ubuntu 19.04
Release:	19.04
Codename:	disco

### Behavior
Trying to slice the following (only the dog, not the box):
https://www.thingiverse.com/thing:220609
we get:

```
ASSERT INFO:
Slic3r has encountered an error
Coordinate outside allowed range
```

the same model is sliced properly with the upstream slic3r, i've printed it multiple times.

uniacke1, 2019-06-12:
I'm observing this too, local build of sources on linux.

uniacke1, 2019-06-21:
This problem is caused by the "Rectilinear (filled)" solid infill pattern. Changing it, or altering the "anchor solid infill" percentages to (e.g.) 1mm/1mm makes the bug go away.

supermerill, 2019-07-02:
can't reproduce with the last version.
If you can, please link me the project file (.3mf)

uniacke1, 2019-07-02:
Sure, here you go.

[FilamentPM-silver.3mf.gz](https://github.com/supermerill/Slic3r/files/3349445/FilamentPM-silver.3mf.gz)

supermerill, 2019-07-02:
Seems like I have fixed this in my dev build.

uniacke1, 2019-07-02:
Superb, I'll pull and recompile to give it a sniff test. Thanks!

uniacke1, 2019-07-03:
This does indeed appear to have been fixed!

VanessaE, 2021-01-27:
This is not fixed, or perhaps I should say it has resurfaced.  I'm at commit a4699e72fc5afbde8506aea8306ad4b9e9ba9970, newly built just before this comment.  Try this project:

[Benchy_Bernie_1.5x.3mf.zip](https://github.com/supermerill/SuperSlicer/files/5880111/Benchy_Bernie_1.5x.3mf.zip)

The model slices fine at 100% (normal) size, but scale it up some and it throws "Coordinate outside allowed range" as per this issue's title after spending some time slicing.

Setting the anchors to 0/0 to disable them did not help, however, setting the solid infill to "rectilinear" (not "filled") per your suggestion *does* work.

jhoblitt, 2021-02-14:
I'm seeing this as well and I can't seem to escape it by changing the infill type(s).

jhoblitt, 2021-02-14:
The only way I can resolve the error is by scaling the part down to 25% scale. Anything much larger then that fails.  This is really odd as previous versions of the part were slicing OK (STLs generated by f360).  I have tried running an STL repair and reimporting the part from a OBJ with no change in behavior.  This is happening with 2.3.55.2.

jhoblitt, 2021-02-17:
@supermerill This is still happening with the nightly build from ~24 days ago.  PrusaSlicer 2.3.0 is able to slice the files.  I started trying older versions and it works with SS 2.3.55.0 but fails with the same error under 2.3.55.1.

tecnopolis-ve, 2021-03-04:
With superslicer 2.3.55.5 does ocurr the same.

I've figured out how to "alleviate" this issue by changing Anchor solid infill by X mm from rectilinear (filled) to rectilinear.

I've tried with a 3dbenchy (normal size does work, scaling over 200% fails)

I think that this issue must be reopened.

TimbreDesign, 2021-03-05:
I have also now experienced this with 2.3.55.

I have been oddly able to stop the error by flipping the print over (requiring support) and lowering the layer height.

Here is the file:

[filament_tube_fix.zip](https://github.com/supermerill/SuperSlicer/files/6090898/filament_tube_fix.zip)

supermerill, 2021-06-02:
can't reproduce in 2.3.56.1

Majestry, 2021-07-03:
I can reproduce this issue on 2 separate models.
Version 2.3.56.6, MacOs BigSur 11.4 (M1 CPU)
Bed size is set to 235x235x250, model size is 85x134x100. The object was positioned automatically, I didn't moved it on the bed.
If i set infill to 0 - model is sliced, otherwise i got error Coordinate outside allowed range
Github do not allow me to download the m3f file (return the 422 with size is not provided, which is weird), so here is the direct link to 3mf file: https://drive.google.com/file/d/18GI9keV177xMhC34SFBrUKOABO-hvkMV/view?usp=sharing

supermerill, 2021-07-03:
@Majestry 
I loaded your .3mf and it slices perfectly on a fresh 2.3.56.6
can you take a screenshot of what's going on after you load & slice it?

Majestry, 2021-07-03:
@supermerill sure, here is the link to screen record when error occures. https://www.youtube.com/watch?v=WMaPeugcHRo
Also, here i loaded the same model and set infill to 0 - and it was sliced without error. https://youtu.be/oSMP38DvXp8

nogginSQZ, 2021-07-08:
I have same issue with another complicated model: also during "Preparing infill". Sometimes, I can work around the problem by fiddling with Slicing -> Filtering -> Resolution (0.0001 or 0.00001 or 0). But otherwise, I'm stuck.

nogginSQZ, 2021-07-09:
I found out more. It turns out that if I set Perimeters & Shell -> Vertical shells -> Perimeters to 1, then my own model produces the "Coordinate outside allowed range" error.  However, by setting the sparse infill to 0%, the error goes away. I can reproduce this behavior with @Majestry's model from a virgin state:
```tmp_dir=$(mktemp -d -t ci-XXXXXXXXXX); ~/Software/SuperSlicer-ubuntu_18.04-2.3.56.6b.AppImage --datadir ${tmp_dir}; rm -rf ${tmp_dir}```
where I click the configuration menu away and then load `~/Downloads/Nezuko_Kamado.3mf`
This model slices fine unless I set number of perimeters to 1. Hope this helps

supermerill, 2021-07-09:
I'll try on my ubuntu vm.

nonchip, 2021-07-19:
just happened to me too, during "Generating perimeters" on this project:
[cube_hologram.zip](https://github.com/supermerill/SuperSlicer/files/6839973/cube_hologram.zip) (renamed 3mf file to zip to shut github up).
I'm on SuperSlicer_2.3.56.6b (installed from [your release](https://github.com/supermerill/SuperSlicer/releases/tag/2.3.56.6b) via appimage)

soupkitchenfilms, 2021-07-21:
This is happening for me on a M1 Mac with Big Sur on version 2.3.56.7
It seems to be only for very high detailed models. It won't slice this moon model without erroring. https://www.thingiverse.com/thing:3532973

supermerill, 2021-08-23:
@nonchip Can't reproduce on 2.3.56.8 on windows.
@soupkitchenfilms Can't reproduce on a prusa mk3 0.2mm, Can you test again on the 2.3.56.8? there is some fix inside it. If it continue to fail, can you post here the failing project file?

RealDeuce, 2021-08-29:
Occured on FreeBSD build of 2.3.56.8 with thing::4717095 scaled to 250mm in largest dimension, lain on one of the backs or on the lower back and cut flat with cut tool, 10% gyroid infill, filled Solid/Top/Bottom.

RealDeuce, 2021-08-30:
[BearHead-bad.zip](https://github.com/supermerill/SuperSlicer/files/7076119/BearHead-bad.zip)
Failed project file.

supermerill, 2021-08-30:
> Failed project file.

Does it also fails if the slicer is locked on one thread/core ?

success on windows.
will try on linux when i'll have the time.

RealDeuce, 2021-08-31:
Yes, it still fails when pinned to a single core.

RealDeuce, 2021-08-31:
Attached the trace level log.  FWIW, it ends with this:
```
[2021-08-31 02:31:58.184419] [0x0000000806849000] [info]    Discovering vertical shells...
[2021-08-31 02:31:58.184441] [0x0000000806849000] [trace]   discover_horizontal_shells()
[2021-08-31 02:32:50.216428] [0x0000000806849000] [info]    Bridge over infill...
[2021-08-31 02:32:53.981397] [0x0000000806849000] [info]    overextrude over Bridge...
[2021-08-31 02:32:53.981406] [0x0000000806849000] [info]    overextrude over Bridge...
[2021-08-31 02:32:53.981410] [0x0000000806849000] [info]    overextrude over Bridge...
[2021-08-31 02:32:53.981413] [0x0000000806849000] [info]    overextrude over Bridge...
[2021-08-31 02:32:53.981420] [0x0000000806849000] [debug]   Filling layers in parallel - start
```
[slice-log.zip](https://github.com/supermerill/SuperSlicer/files/7081761/slice-log.zip)

Using a normal for() instead of tbb::parallel_for() fails in the same manner.

Printing the values gives me this:
Coordinate outside allowed range... -9223372036854775808, -9223372036854775808 > 4611686018427387903

Which, -9223372036854775808 is 0x8000,0000,0000,0000

And now I'm off to bed. :)

supermerill, 2021-08-31:
can reproduce on linux

supermerill, 2021-09-04:
Found the problem.
Seems like on linux/g++ boost::voronoi can return nan
Also the std::abs on g++ doesn't keep nan and write a big number instead.

Belstrekkie, 2021-11-13:
Both version 2.3.57.4 and 2.3.57.5 are giving me the coordinate outside allowed range in the Windows version, whereas I haven't had any issues in 2.3.57.3 and earlier. In the image the sliced object is version .3 and the error .4. The position and settings are exactly the same.
[image]

alfrix, 2021-11-13:
i have also found this bug in windows 2.3.57.4 and 2.3.57.5, but not on 2.3.57.3 (same as Belstrekkie)

adding a cube and moving it to the left past 250 mm produces the error
[image]

[Aspecto-Caja.zip](https://github.com/supermerill/SuperSlicer/files/7532646/Aspecto-Caja.zip)

Maximilian-Staab, 2021-11-14:
I'm getting the same error on 2.3.57.5, but only if "brim" is enabled:
[image]

Maximilian-Staab, 2021-11-18:
The nightly build fixed the issue for me, thanks!

closed 2021-12-15 by supermerill

Dreistein75, 2021-12-20:
Sorry, not good with GitHub. I have the newest SuperSlicer (MacOS on ARM) and the following STL from PrusaPrinters gives the Error with the "Coordinate outside allowed range". So guess this is still not fixed entirely.

[white-christmas.stl.zip](https://github.com/supermerill/SuperSlicer/files/7741961/white-christmas.stl.zip)

supermerill, 2021-12-20:
@Dreistein75 
can't reproduce. Maybe post your project file?

Dreistein75, 2021-12-20:
@supermerill 
I just re-tried and the error is still there. I saved the project as 3mf and I will attach it here. It is SuperSlicer 2.3.57.7 for ARM MacOS

[SuperSlicerProblem.3mf.zip](https://github.com/supermerill/SuperSlicer/files/7745737/SuperSlicerProblem.3mf.zip)

INeDiA, 2021-12-21:
> @supermerill I just re-tried and the error is still there. I saved the project as 3mf and I will attach it here. It is SuperSlicer 2.3.57.7 for ARM MacOS
> 
> [SuperSlicerProblem.3mf.zip](https://github.com/supermerill/SuperSlicer/files/7745737/SuperSlicerProblem.3mf.zip)

same here; switched back to the regular release (no ARM) and it works without any issue

supermerill, 2021-12-21:
I think something is creating Nan somewhere when compiled with arm. Next to impossible to fix without a mac mini on hand. Problem is, they are still kind of expensive.

yonkiman, 2021-12-22:
I'm also getting a"Coordinate outside allowed range" error when I try to move the object towards the upper right of the build plate.  If I pull it in closer to the center it slices fine.  I've attached the project file
[Coordinate outside allowed range SS 2.3.57.5, Windows 10.zip](https://github.com/supermerill/SuperSlicer/files/7765231/Coordinate.outside.allowed.range.SS.2.3.57.5.Windows.10.zip)
.

supermerill, 2021-12-22:
@yonkiman on the 2.3.57.5 yes, but it doesn't happen on the latest release (tested on 2.3.57.8)

yonkiman, 2021-12-23:
Apologies - should have made sure I was on the latest version!

supermerill, 2022-07-12:
@elcojacobs no problem for me.
your project?

elcojacobs, 2022-07-13:
This is very strange, I cannot reproduce the error now. It happened on multiple models, even after restart of the appimage.
If it ever happens again, I'll get you a 3mf.

elcojacobs, 2022-07-13:
Seems fixed in yesterdays nightly. Could you reproduce it in the 'better fixes' nightly?
The bug with 2 times the same checkbox label is still there.