supermerill/SuperSlicer #67: print every second layer at opposite part order

burner-, 2019-05-11:
Currently if I put 2 parts to plater it will print them with looping order. Layer 1 for part 1, layer 1 for part 2, layer 2 for part 1, Layer 2 for part 2. 
That will cause travel, and seaming each time when layer is changed. 
If part is printer layer 1 for part 1, layer 1 for part 2, layer 2 for part 2, layer 2 for part 1 it will save travels and perimeter crossings.

neophyl, 2020-08-28:
Except in the use case where you have tall thin parts that the user has specifically added extra parts to allow the others time to cool before depositing the next layer on.

n8bot, 2020-08-28:
Yeah, this is often detrimental to prints. Small islands can end up having too much heat put into them because they are the last to get printed on one layer, and the first to get printed on the next layer.

It would have to be an option to turn on or off. This is how simplify3d handles it. They have a Boolean option: "print islands without optimization" or something to that effect.