supermerill/SuperSlicer #68: Can' import STL on 1.42.0-alpha7.0_190314 but OK on 1.42.0-alpha6.0_190221

Serveurperso, 2019-06-03:
### Version
_Version of Slic3r++ used goes here_
Slic3r++_1.42.0-alpha7.0_190314

_Use `About->About Slic3r++` for release versions_
N/A

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_
Last 64 bits binary downloaded here

### Operating system type + version
_What OS are you using, and state any version #s_
Windows 10

_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_
MK3S

### Behavior
* _Describe the problem_
Crash when importing STL
Can' import STL on last version 1.42.0-alpha7.0_190314 
It work OK on previous version 1.42.0-alpha6.0_190221

* _Steps needed to reproduce the problem_
File -> Import -> Any STL

  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
Opening STL

* _Actual Results_
  * _Screenshots from __*Slic3r++*__ preview are preferred_
Crash
Console :
```
System OpenGL library released
Loading MESA OpenGL library: C:\Users\Pascal\Desktop\Slic3r++_1.42.0-alpha7.0_190314\mesa\opengl32.dll
MESA OpenGL library was loaded sucessfully
Multisample disabled
Something is syntactically very wrong with this ASCII STL!: No error
```
[image]

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
_Upload a Slic3r++ Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for Slic3r++, Slic3r PE 1.42.0-alpha and newer)_

supermerill, 2019-06-05:
Can you try to import the stl in Microsoft 3D builder and export it before importing it in slic3r?

Serveurperso, 2019-06-05:
OK I test this soon.
Another problem : On working previous version, there is no calibration sequence like the official PrusaSlicer, I can't use it.

Serveurperso, 2019-06-18:
Hi, sorry for the delay,
Any STL I export from any software do the problem.
Here is a scad file with a STL with the problem

[plate-with-holes.zip](https://github.com/supermerill/Slic3r/files/3301566/plate-with-holes.zip)

closed 2019-07-08 by supermerill