supermerill/SuperSlicer #72: Missing MSVCP140.dll

neophyl, 2019-06-27:
### Version
_Version of Slic3r++ used goes here_
1.42.0 win 64 260609 version 

### Operating system type + version
Windows 7

### Behavior
Tried to run the exe just like I do for all your other previous versions but get an error about missing dll.
[image]

supermerill, 2019-06-27:
I have built it with vs2017 instead of vs2013. Maybe some distribuable missing.
I think i forgot to replace the xxxx120.dlml by the 140 version, and as i have everything install on my dev computer, i missed that.

in the meantime, you can use ithis release by installing the Microsoft visual c++ distribuable for visual studio 2017 : https://aka.ms/vs/15/release/VC_redist.x64.exe (or from the download page : https://visualstudio.microsoft.com/downloads/)

neophyl, 2019-06-27:
Yeah I already grabbed the 2015 redistributable so I was already playing with it at work earlier but just thought I should let you know a dll was missing.  Looks like the Plicer 2.0 ships with the 120 rather than the 140 so I couldn't just copy it across.

Anyway thanks for all your work doing these.  Time to play some more :)

closed 2019-07-31 by supermerill