supermerill/SuperSlicer #73: Brim ears

reloxx13, 2019-06-28:
### Version
Slic3r++_1.42.0_260609

### Operating system type + version
Win10x64

### Behavior
Brim ears not generated with atest update.

Tried diffrent Models and Degrees, the ear brims does not show up :/

#### Project File (.3MF) where problem occurs

[Raspberry_Pi_4_Case_-_Top_12_With_30_mm_fan_Mount_and_GPIO_cut_1.zip](https://github.com/supermerill/Slic3r/files/3339865/Raspberry_Pi_4_Case_-_Top_12_With_30_mm_fan_Mount_and_GPIO_cut_1.zip)

supermerill, 2019-06-29:
This little feature can only work on sharp edges, and not on rounded ones.

reloxx13, 2019-06-29:
hmm you are right. i tried again with a sharp corners model and the ears appeared.
   
but the previously printed lower part of the case showed up the ears (and it has round corners, too. now if i put the lower part in again it does not show the the ears, even not with the previous RC version (downloaded it again) which i used before to print the lower case. strange.

---
lol look what happened now :D

[image]

supermerill, 2019-06-30:
when that happen, save the project (.3mf) and put it here, to enable me to correct the bug.

supermerill, 2019-07-02:
A little tip, because your case has mostly strait section, you can have ears with a max angle to 178°.

I will update/upgrade the tooltip a bit.

reloxx13, 2019-07-02:
oh dude, i did not try 180° since 180 is not a edge :D
gone up between 10 and 145

closed 2019-07-02 by supermerill