supermerill/SuperSlicer #75: Feature request: Gyroid tweak

uniacke1, 2019-07-02:
This is a future feature request to include the gyroid improvement shown here:
(note: Not currently available, I'm just marking it for the record so we can hopefully suck it in in the future and make one more featureful fork than 30 different forks each with one feature. But I understand if you don't want to deal with maintaining and importing someone else's code)

https://www.youtube.com/watch?v=SMEVg4qGzEI

wavexx, 2019-07-02:
Whoa, I didn't think I would see this!

The code is obviously public for this, there are three commits currently which are based up on this code anyway (see https://github.com/prusa3d/PrusaSlicer/issues/1690 where this started).

- https://github.com/wavexx/PrusaSlicer/commit/8d06c87ea19e0bd607014dacda580fc2ba0cc566
- https://github.com/wavexx/PrusaSlicer/commit/a73d625638ec6cbc0d8b64b589c3093ccd7d917e
- https://github.com/wavexx/PrusaSlicer/commit/aa0eed82cbfd015e6a5dea232b8178c925e89775

I didn't do a PR yet as I don't think the current changes are ready. Layer thickness still needs to be taken into account. And as I rebased frequently on top of prusa's master, the commit hashes changed a few times..

uniacke1, 2019-07-02:
We're 3d printing nerds, we like our features bleeding edge, and often :D

supermerill, 2019-07-02:
> the commit hashes changed a few times

Oh, that's why i'm notifdied every day about the same commit!

I'll wait your release to merge.

wavexx, 2019-07-02:
Sorry for this, but I try to limit the number of branches I have, otherwise I really have an hard time keeping up as well :/

supermerill, 2019-07-02:
no worry, do as you want.

wavexx, 2019-08-08:
Didn't forget!
https://github.com/wavexx/PrusaSlicer/tree/gyroid_improvements or 
https://github.com/prusa3d/PrusaSlicer/pull/2730

I'm not going to rebase this :)

closed 2019-10-18 by supermerill