supermerill/SuperSlicer #76: Linux compile fixes for G++7

uniacke1, 2019-07-03:
Submitting two minor compile fixes to address things `G++` errors on but VS doesn't by default. 

### Version
Current master, Jul 3/2019
1.42.0+0-6-g4cc1a1fd1

### Operating system type + version
Linux, OpenSusE 42.3, Boost 1.66, wxwidgets 3.0, GCC/G++ 7

[linux-patch.diff.gz](https://github.com/supermerill/Slic3r/files/3355008/linux-patch.diff.gz)

```
diff --git a/src/slic3r/GUI/3DScene.cpp b/src/slic3r/GUI/3DScene.cpp
index cd747eb24..eb1c77bd0 100644
--- a/src/slic3r/GUI/3DScene.cpp
+++ b/src/slic3r/GUI/3DScene.cpp
@@ -1670,7 +1670,8 @@ void ExtrusionToVert::use(const ExtrusionEntityCollection &collection) { for (co
 
 void _3DScene::extrusionentity_to_verts(const ExtrusionEntity &extrusion_entity, float print_z, const Point &copy, GLVolume &volume)
 {
-    extrusion_entity.visit(ExtrusionToVert(print_z, copy, volume));
+    Slic3r::ExtrusionVisitorConst tmp = ExtrusionToVert(print_z, copy, volume);
+    extrusion_entity.visit(tmp);
 }
 
 void _3DScene::polyline3_to_verts(const Polyline3& polyline, double width, double height, GLVolume& volume)
diff --git a/src/slic3r/GUI/Tab.cpp b/src/slic3r/GUI/Tab.cpp
index f4127cd37..b1fab99dc 100644
--- a/src/slic3r/GUI/Tab.cpp
+++ b/src/slic3r/GUI/Tab.cpp
@@ -923,7 +923,7 @@ void Tab::update_preset_description_line()
                _(L("It's a system preset.")) : 
                wxString::Format(_(L("Current preset is inherited from %s")), (parent == nullptr ? 
                                                                                                        _(L("default preset"))+"." : 
-                                                                                                       ":\n\t" + parent->name));
+                                                                                                       _(L(":\n\t")) + parent->name));
        
        if (preset.is_default || preset.is_system)
                description_line += "\n\t" + _(L("It can't be deleted or modified. ")) + 
```

supermerill, 2019-07-03:
Can i have the error messages?

For the first one, i may re-create other ones if i don't understand what's wrong.

For the second one, I don't understand, as :\n\t is not something to translate.

uniacke1, 2019-07-03:
Of course. You are right that the second one is the implicit conversion that makes things unhappy. May not be the most elegant way to fix it but I opted for a more compact inline on item 2.

Item one: 
Slic3r/src/slic3r/GUI/3DScene.cpp:1673:28: error: cannot bind non-const lvalue reference of type ‘Slic3r::ExtrusionVisitorConst&’ to an rvalue of type ‘Slic3r::ExtrusionVisitorConst’
     extrusion_entity.visit(ExtrusionToVert(print_z, copy, volume));
                            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

Item two:
Slic3r/src/slic3r/GUI/Tab.cpp: In member function ‘void Slic3r::GUI::Tab::update_preset_description_line()’:
Slic3r/src/slic3r/GUI/Tab.cpp:924:84: error: operands to ?: have different types ‘wxString’ and ‘std::basic_string<char>’
   wxString::Format(_(L("Current preset is inherited from %s")), (parent == nullptr ?
                                                                  ~~~~~~~~~~~~~~~~~~^
              _(L("default preset"))+"." :
              ~~~~~~~~~~~~~~~~~~~~~~~~~~~~                                           
              ":\n\t" + parent->name));
              ~~~~~~~~~~~~~~~~~~~~~~

uniacke1, 2019-07-03:
Pulled and verified the changes build. Thanks!

closed 2019-07-03 by uniacke1