supermerill/SuperSlicer #77: Ironing happens before infill; causes heatbreak jams

uniacke1, 2019-07-06:
### Version
1.42.0+0-45-g6b20a930e

### Operating system type + version
Linux, OpenSusE 42.3, Boost 1.66, wxwidgets 3.0, GCC/G++ 7

### 3D printer brand / version + firmware version (if known)
Original Mk3, F/W 3.7.2

### Behavior
When slicing with interrupted top infill (e.g. non regular shapes like a circle, or a partial top layer) the ironing sometimes happens prior to the actual infill. This causes two issues:
1. The layer finish is obviously very uneven from regular infill running overtop the ironing extrusion.
2. The extruder has a tendency to jam from the back pressure of the regular infill; either filament grinding or a backup into the heat break leading to a jam there.

Slice any object with an interrupted top infill. My examples were a cylinder with  a hole in the middle and a rectangular slab with a protruding shape on top. 

* _Expected Results_
* _Actual Results_
Screenshot from pronterface gcode inspection:

[image]

#### Project File (.3MF) where problem occurs
[Ring.3mf.gz](https://github.com/supermerill/Slic3r/files/3364830/Ring.3mf.gz)
[Ring.gcode.gz](https://github.com/supermerill/Slic3r/files/3364833/Ring.gcode.gz)

supermerill, 2019-07-06:
I'm not sure to understand.

What line(s) on the gcode file i need to look at?

problem is that the ironing solid infill is done after rectilinear solid infill?
problem is that the ironing is extruded but the two pass (fill+smooth) are done in reverse?

uniacke1, 2019-07-06:
I attached the gcode for reference so you can step through the travel with e.g. pronterface. 

The problem is that for some regions, the order is reversed, e.g. ironing first and then rectilinear infill.

supermerill, 2019-07-06:
> I attached the gcode for reference so you can step through the travel with e.g. pronterface.

yeah, i've done that.
but where I have to look in the gcode? at least, give me the layer.

uniacke1, 2019-07-06:
OK, I'll do my best:
-> Ironing: last layer infill choice, my understanding is it does a final pass over the top surface extruding just a tiny amount of filament to smooth over the top surface. ("Last" layer, low flow/volume)
-> (Rectilinear) infill: referring to the pattern that makes up the "bulk" of the last layer, which is laid down before the ironing can happen. (normal flow/volume extrusion - I'll refer to this as "second last layer" from now on)

Problem: Using the above notation: "Second last layer" is being placed down *before* the last layer. 
Lets step through the topmost layer in pronterface:

Perimiters. (OK):
[image]

Region of "ironing"/last layer (not OK, should be after all "second last layer" parts):
[image]

Region of infill ("second last"): 
[image]

More "ironing"/last layer, again before all 2nd last layer is complete:
[image]
[image]

More "2nd last layer":
[image]

^^ This last screen is where the 2nd last layer is complete and I would now expect the "ironing" process to start happening. Otherwise, the very low flow ironing blobs up and yeilds very poor surface finish, in addition to the extrusion problems noted before.

Did that help clarify? If not perhaps it will help if you see it "live"; I did attach the 3mf used to produce the gcode and it is a very small part/quick print - watching it happen will probably clarify far more than pages of words will.

supermerill, 2019-07-06:
I understand.

It seems that something has broke the "no sort" flag in a merge with prusa's changes. I'll check.

closed 2019-07-31 by supermerill