supermerill/SuperSlicer #78: Git Repo Name

reloxx13, 2019-07-13:
Rename your git repo to Slic3r++ or PrusaSlicer++ (make a decision or get a new name) and make ur release names and versions constant 👍 

its hard to find your fork.

would be nice to have a ci script building nightly releases or make more beta/alpha releases if u fied important stuff.

supermerill, 2019-09-09:
I tried to rename to Slic3r++ but the '+' isn't allowed by github :-/

domesticatedviking, 2019-09-26:
Slic3r++ evaluates to Slic4r as far as I'm concerned.   Though I was never a big fan of the 'leet '3' in the name.

photog0411, 2020-01-01:
SuperSlic3r
SuperPrusaSlicer
PrusaSlicer Ultimate Edition

sci-phi, 2020-02-19:
Slic3r-Plus ?

combolek, 2020-05-13:
It would be great if we could drop the '3'.
Maybe SlicerPP ?

Photogad, 2020-05-13:
Super Prusa Slicer

guestisp, 2020-05-13:
slic3r-plus-plus like notepad++:
https://github.com/notepad-plus-plus/notepad-plus-plus ?

supermerill, 2020-05-13:
5l1c3r++ to be sure that Thomas Sanladerer can't pronounce it.

good idea for the -plus

neophyl, 2020-05-13:
That’s just cruel.  Funny but cruel 😂

supermerill, 2020-05-23:
I think I'll go with SuperSlicer, so the @Photogad proposal but without the '3'.
And as i add 'super' when my username is already taken, it's fitting.

closed 2020-07-03 by supermerill

cab404, 2020-09-09:
I propose prepending "Legendary" to the name. It will fix "SS" acronym!

supermerill, 2020-09-09:
The new acronym seems to be Susie.
Found it used on a discord.