supermerill/SuperSlicer #79: Crash

userosos, 2019-07-16:
Version: 2.0.0-1
Build: slic3r++_2.0.0-1+UNKNOWN
Operating System: Windows
System Architecture: 64 bit
Windows Version: Windows 10 (build 17763), 64-bit edition
GL version: 4.6.0 NVIDIA 425.45
Vendor: NVIDIA Corporation
Renderer: Quadro P1000/PCIe/SSE2
GLSL version: 4.60 NVIDIA
I have crash everytime after i press slice! I can not use it! :(

supermerill, 2019-07-16:
can i have your project? clic file-> save as instead of clicking slice. Then zip the file and drag-top here.
With it, i may be able to resolve the bug.

MissingNoIOI, 2019-07-25:
It crashes for me too, I've narrowed it down to the support generation. With supports set to "nowhere" it slices fine, but once I enable them, doesn't matter if anywhere or just on the bed, it crashes during slicing. I've tried it with 3 models so far and all have crashed, here is one of them:
[Bed_Direction.zip](https://github.com/supermerill/Slic3r/files/3429215/Bed_Direction.zip)

Here are my settings:
[Slic3r++.zip](https://github.com/supermerill/Slic3r/files/3429223/Slic3r%2B%2B.zip)

I'm also on Windows 10 Pro (build 1803) 64bit, but I'm using an AMD graphics card

MissingNoIOI, 2019-07-25:
Short addition, I tested it again in 1.42.0, it slices without a problem there, only 1.42.1 crashes

closed 2019-07-31 by supermerill